ETLCPP/etl #290: Problem with etl::vector::emplace_back

r2r0, 2020-10-21:
Code below does not compile, where similar code based on std::vector compiles (and works) without any problem.

struct SomeObject {};
etl::vector<SomeObject*, 10> vec {};
vec.emplace_back(nullptr);

To avoid problem I have to define simple wrapper i.e.
struct SomeObjectWrapper
{
    SomeObjectWrapper(SomeObject* a_ptr) : ptr(a_ptr) {}
    SomeObject* ptr;
};
etl::vector<SomeObjectWrapper, 10> vec {};

Is it intentional behavior?

Compiler: GCC 9.3.1 on OpenSUSE

jwellbelove, 2020-10-21:
No, it's not intentional, I've just never tested for that scenario.
I'll take a look to see what can be done.

jwellbelove, 2020-10-21:
It seems that I neglected to give the `etl::vector` specialisation for pointers an `emplace_back` implementation.
I'll raise an issue for this.

jwellbelove, 2020-10-21:
Can you use push_back for the moment?

r2r0, 2020-10-21:
Thanks for promp follow up.
No problem, for such basic type push_back works well.

jwellbelove, 2020-10-21:
Fixed in release 18.17.2

closed 2020-10-21 by jwellbelove