ETLCPP/etl #292: Infinite loop in etl::map find_limit_node()

Hyvok, 2020-10-23:
I'm having some weird issue with etl::map with a type etl::map<etl::string, etl::delegate> for mapping some shell command strings to lambdas.

I am doing something like trying to list all the registered commands (commands is the etl::map):
```c
commands["help"] = [&](PrintFn print, const etl::string_view) {
    for(const auto& cmd : commands)
    {
        print(cmd.first);
    }
    return 0;
};
```
But the end result is that the the for-each loop the map ends up in an infinite loop in etl/include/map.h find_limit_node() where dir = 0, position = 0xA5A5A5A5 and limit_node seems to always stay as = 0xA5A5A5A5. I'm probably doing something wrong but cannot see what would be causing this. Any ideas?

jwellbelove, 2020-10-23:
OK, I don't _think_ you are doing anything wrong, but could you try the `for` loop outside of the lambda?

Hyvok, 2020-10-23:
Outside the lambda looping over the commands map seems to work as expected.

jwellbelove, 2020-10-23:
Is it possible that `commands` has gone out of scope when the lambda is called?

Hyvok, 2020-10-23:
Should not be, the commands map is a private member of my Shell class. Shell class itself calls the associated commands via the map when a command string is found in the input to the shell.

I also tried capturing this pointer in the lambda and accessing commands via this->commands but it had no effect on the behavior.

jwellbelove, 2020-10-23:
Have you tried a breakpoint in the lambda to see if the `commands` map appears to contain the same data as the outer container?

jwellbelove, 2020-10-23:
Also, do you get the same problem if you change the lambda to a real function?

Hyvok, 2020-10-23:
By changing the lambda to map to a member function of the Shell class seems to work fine!

As for the first question not quite sure what am I looking for, if I do "print &commands" in the input processing function which is a member of the Shell instance I get the type + "0x5b79c <ucHeap+16176>" (my project is using FreeRTOS and the task where everything is happening is located on the heap) and if I do the same in the lambda context I get the type + "0xa5a5a625".

jwellbelove, 2020-10-23:
As your lambda is capturing by reference, I would have expected the addresses of the outer and lambda maps to be the same.
What happens if you try to print size()?

Hyvok, 2020-10-23:
If I try to do "call commands.size()" gdb says "cannot evaluate function -- may be inlined". But yeah if I "print commands" in lambda scope everything seems to be 0xA5A5A5A5 (root_node, current_size etc.) which I am guessing is uninitialized memory... Wonder why is this happening.

jwellbelove, 2020-10-23:
Definitely sounds like uninitialised memory.

Hyvok, 2020-10-23:
Ok now I have a minimal example that causes address-sanitizer error in my test build:
```cpp
    std::string str("bur");
    etl::map<etl::string<8>, etl::delegate<void()>, 4> map;
    map["foo"] = [&]() {
        str.push_back('B');
    };
    auto it = map.find("foo");
    if(it != map.end())
    {
        it->second();
    }
```
This generates the attached address-sanitizer output (ignore some catch stuff from the testing framework).
[asan.txt](https://github.com/ETLCPP/etl/files/5430010/asan.txt)

jwellbelove, 2020-10-23:
Ah, you're using etl::delegate. One limitation of etl::delegate is that it cannot be used with captured variables, if the variables are out of scope when it is called. I suspect that is what is happening here somehow.

I have to admit that I am not a lambda 'guru'.

jwellbelove, 2020-10-23:
Could it be that the reference to `commands` is stored on the lamba's stack frame, which is then out of scope when called. (I'm guessing a bit here!)

jwellbelove, 2020-10-23:
Does this help at all? It talks about capturing `this`.

https://stackoverflow.com/questions/23803152/why-cant-a-data-member-be-in-a-lambda-capture-list

Hyvok, 2020-10-23:
Capturing this and trying to access via "this->commands" does not seem to help, this has a value of 0xA5A5A5A5 in the lambda scope again.

I think the earlier minimal example shows the same issue, but here is my specific situation in terms of what is in scope:
```cpp
class Shell
{
    Shell()
    {
        commands["help"] = [&]() {
            for(const auto& cmd : commands)
            {
                // Print commands
            }
        };
    }
    void input(const std::string& inStr)
    {
        auto it = commands.find(inStr);
        if(it != commands.end())
        {
            it->second();
        }
    }

private:
    etl::map<etl::string, etl::delegate<void()>> commands;
};

int main
{
    Shell shell;
    // Task event loop
    for(;;)
    {
        shell.input("help");
    }
}
```

jwellbelove, 2020-10-23:
Thanks, I'll have a play with it here and see what I can find out.

Hyvok, 2020-10-23:
Definitely an etl::delegate issue, the following fails with address-sanitizer error too:
```cpp
    std::string str("foo");
    etl::delegate<void()> fn = [&]() {
        str.push_back('B');
    };
    fn();
```

jwellbelove, 2020-10-23:
I tried adding your code to the `etl::delegate` unit tests, but every thing is fine. All tests pass.

```
    //*************************************************************************
    TEST_FIXTURE(SetupFixture, test_issue_292)
    {
      std::string str("foo");
      etl::delegate<void()> fn = [&]() 
      {
        str.push_back('B');
      };

      fn();

      CHECK_EQUAL("fooB", str);
    }
```

Hyvok, 2020-10-23:
Wonder if this could be some compiler thing then... Though I am using a different version of GCC for my tests and embedded code. For the tests I am running "gcc version 8.4.0 (Ubuntu 8.4.0-1ubuntu1~18.04)" and for the embedded I am running "gcc version 9.2.1 20191231 (Altera 20.1 Build 711)".

For the tests address-sanitizer settings (via environment variable ASAN_OPTIONS) I am running the following extra options: `strict_string_checks=1:detect_stack_use_after_return=1:check_initialization_order=1:strict_init_order=1`

And for the actual compiler flags I have:
```cmake
# Instrument the code with test coverage checking
target_compile_options(${PROJECT_NAME} PUBLIC
    -fprofile-arcs
    -ftest-coverage
    -fprofile-abs-path
    -fsanitize=address
    -fsanitize=pointer-compare
    -fsanitize=pointer-subtract
    -fsanitize=undefined
    # Makes debugging output from the sanitizer easier to read
    -fno-omit-frame-pointer
)
```

jwellbelove, 2020-10-23:
The unit tests on my machine are using VS2019 & GCC 8.1.0

closed 2020-11-13 by jwellbelove