ETLCPP/etl #296: #include string.h caused errors from flags.h

lfenster, 2020-11-03:
was trying this library out and testing using strings capability.  #include "string.h" was throwing a lot of errors from flags.h.  seemed to be due to constref.  backing out latest version and going back to 18.18.2 for PIO resolved these errors.

also, is to_upper_case gone?

jwellbelove, 2020-11-03:
Did you mean errors from 'constexpr'?

jwellbelove, 2020-11-03:
Can you tell me what compiler & C++ standard you are using?

Can you post a minimal example that produces the error?

jwellbelove, 2020-11-03:
Yes, to_upper_case, to_lower_case, to_sentence_case were removed as they were not supported for `wstring`, `u16string` & `u32string`.
In retrospect, I should probably still make them available for `string`.

lfenster, 2020-11-03:
Yes, sorry, that is what I meant

lfenster, 2020-11-03:
Thanks John.  Understood.  It would be awesome if it could still be supported for string.  BTW, I absolutely love this library.  Really nice work.  Thank you for providing it!

lfenster, 2020-11-03:
right now I am using PIO and native; gcc (mingw) compiler.  On Windows OS but configured in PIO to be native.  Is that what you needed, John?

jwellbelove, 2020-11-03:
C++03, C++11 or higher?

lfenster, 2020-11-03:
good questions.  C++11 (I believe).  below is the platform.ini for the native part. Does this help?

[env:nativetest]
platform = native
build_flags = 
	-std=gnu++11
	-D PROFILE_GCC_GENERIC
	-I include
lib_extra_dirs = ~/Documents/Arduino/libraries
lib_deps = 
	fabiobatsilva/ArduinoFake @ ^0.2.2
	etlcpp/Embedded Template Library@18.18.2

jwellbelove, 2020-11-03:
I see you have -D PROFILE_GCC_GENERIC. Are you using the CMakefile.txt? If so, things like PROFILE_GCC_GENERIC have not existed in that file since the end of September, and didn't do anything much for some time before that.

Can you look in platform.h and see which sections of code are enabled? If GCC with C++11 current then ETL_COMPILER_GCC should be defined and ETL_CPP11_SUPPORTED enabled, ETL_CPP14_SUPPORTED and ETL_CPP17_SUPPORTED disabled.

Do you have a user etl_profile.h defined?

Can you post an actual error message?

lfenster, 2020-11-03:
thanks John. I removed the -D PROFILE... and it tested okay.  When I look in platform.h, it looks like none of them are supported but i actually have to environments and i dont know if that is throwing it off in the IDE.  the tests under native are working now.  However, I now the build for Arduino is failing for what seems like something missing in the etl_profile.h for it as far as I can guess.  I'm not certain; configuring the etl_profiles properly feels a little like an acquired art form to me.

providing more info to hopefully 1) help you identify the challenge i had as well as 2) hope you can help me see what i am doing incorrectly for the arduino environment.
The _platform.ini is:_ file contains
[platformio]
default_envs = ShadowTubeCeiling

[env:ShadowTubeCeiling]
platform = atmelavr
board = megaatmega2560
framework = arduino
build_flags = 
	-I include
	-I include/ShadowTubeCeiling
lib_extra_dirs = ~/Documents/Arduino/libraries
lib_deps = 
	arduino-libraries/Servo @ ^1.1.7
	etlcpp/Embedded Template Library@18.18.2

[env:nativetest]
platform = native
build_flags = 
	-std=gnu++11
	-I include
	-I include/nativetest
lib_extra_dirs = ~/Documents/Arduino/libraries
lib_deps = 
	arduino-libraries/Servo @ ^1.1.7
	fabiobatsilva/ArduinoFake @ ^0.2.2
	etlcpp/Embedded Template Library@18.18.2

The _include\nativetest\etl_profile.h_ file contains
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__

#define ETL_THROW_EXCEPTIONS
#define ETL_VERBOSE_ERRORS

#if defined(_MSC_VER)
  #include "etl/profiles/msvc_x86.h"
#else
  #include "etl/profiles/gcc_windows_x86.h"
#endif

#endif

The _include\ShadowTubeCeiling\etl_profile.h_ file contains
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__

#define ETL_ARDUINO_INCLUDED
#define ETL_TARGET_DEVICE_ARM
#define ETL_TARGET_OS_NONE
#define ETL_COMPILER_GCC
#define ETL_CPP11_SUPPORTED 0
#define ETL_CPP14_SUPPORTED 0
#define ETL_CPP17_SUPPORTED 0
#define ETL_NO_NULLPTR_SUPPORT    1
#define ETL_NO_LARGE_CHAR_SUPPORT 1
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED 0
#define ETL_NO_STL 1

#endif

the tests run under native are working okay now.  however, there is a reference to #include "etl/string.h" in one of my header files that is now causing the ShadowTubeCeiling (arduino) environment not to build with a lot of errors.  Ive played around with a lot of different configs but cannot seem to get something working now.  If I remove the define for ETL_NO_STL  then I just get _.pio\libdeps\ShadowTubeCeiling\Embedded Template Library\include/etl/utility.h:41:25: fatal error: algorithm: No such file or directory_ but if ETL_NO_STL is included as above, there's a whole bunch of errors, starting with _.pio\libdeps\ShadowTubeCeiling\Embedded Template Library\include/etl/algorithm.h:870:70: error: macro "min" passed 3 arguments, but takes just 2_; both emanating from the etl/string.h at line 35 to basic_string.h line 40 (which calls algorithm.h)

thanks again for all your help with this.  hopefully, this is provide some insight for you into the issue?
thanks
Lenny

lfenster, 2020-11-05:
im still unsure what is going on.  In the native tests (windows), everything builds and tests okay.  However, in the arduino build, I keep getting a _.pio\libdeps\arduino\Embedded Template Library\include/etl\type_traits.h:69:25: fatal error: type_traits: No such file or directory_. I've commented out code and it seems to either originate from _strings.h_ or _vector.h_; both of which call _basic_string.h:40_ then _algorithm.h:43:0_ and then _type_traits.h:69:25_.  any ideas??

jwellbelove, 2020-11-05:
As you are compiling for C++11, the ETL is expecting the STL  <type_traits> header to be available.
This doesn't seem to exist for you. That's the common problem with Arduino; it doesn't contain everything you would expect from a C++ platform.

lfenster, 2020-11-05:
Thanks @jwellbelove .  I tried with ETL_NO_STL in etl_profile but still no luck.  Is there some other suggestions to get past this error?  I’d really love to keep leveraging this library

jwellbelove, 2020-11-05:
I assume with ETL_NO_STL you are not getting the `<type_traits>` error anymore, just the `constexpr` issues?

The only thing I can think of at the moment is to disable constexpr by adding  `&& !defined(ARDUINO)` to the C++11, C++14, C++17 & C++20 sections of `platform.h`.

```
// The macros below are dependent on the profile.
// C++11
#if ETL_CPP11_SUPPORTED && !defined(ETL_FORCE_NO_ADVANCED_CPP) && !defined(ETL_ARDUINO)
```

lfenster, 2020-11-05:
i think you're right about type_traits. With NO_STL I am seeing different errors but I went back to 18.19.1 to see if it would help so it may be due to that.  the errors i see with NO_STL are all from limits.h:
_'nan' was not declared in this scope
'nanf' was not declared in this scope
'nanl' was not declared in this scope
'HUGE_VALF' was not declared in this scope
'HUGE_VAL' was not declared in this scope
'HUGE_VALL' was not declared in this scope_

jwellbelove, 2020-11-05:
I've just looked back at the Git history and it seems there was a PR in October for fixing that, but it seems that I haven't merged it back into master yet!

lfenster, 2020-11-05:
😀how can I help?  Would it help for me to test with a different branch?  Just let me know what I can do to help with this @jwellbelove 

Thanks
Lenny

jwellbelove, 2020-11-13:
These should all be fixed in v18.20.0

closed 2020-11-13 by jwellbelove