ETLCPP/etl #297: Keil 5 (armcc 5) compilation issues.

rolan-reznik, 2020-11-03:
Hello,
There are some armcc5 compilation issues with version 18.19.2 when c++11 profile is enabled.

1. https://github.com/ETLCPP/etl/blob/65fa8c51e4b47c4d36ab982cb31cbc65af5e27b2/include/etl/type_traits.h#L754
   Previously there was a fix for armcc5:
   ```
   #if ETL_CPP11_SUPPORTED && !defined(ETL_COMPILER_ARM5)
   ```
   now it is gone.

 2.  `basic_string.h` fails to compile due flags.h with following errors:
     ```
      "statement may not appear in a constexpr function" value ? data |= (pattern & MASK) : data &= (~pattern & MASK)
      ```
      Probably it is due limited (buggy) `constexpr` support for armcc5.

3. There is a warning for function.h in line:
    https://github.com/ETLCPP/etl/blob/65fa8c51e4b47c4d36ab982cb31cbc65af5e27b2/include/etl/function.h#L185
    because `virtual` keyword is missing.

jwellbelove, 2020-11-04:
OK, I'll look at those today.

rolan-reznik, 2020-11-04:
Thank you very much. I will be able to test fixes this week.

lfenster, 2020-11-04:
+1.  I was seeing this as well based on what i was using.  I thought it was just me though

rolan-reznik, 2020-11-04:
Clarification about flags `constexpr` problem. I gave only one error example. There was alot of errors in all places were `constexpr` method was using ternary operator.

lfenster, 2020-11-04:
I saw the same and think it may be related the ‘strings.h’ issue I raised just a few days ago.  I had to roll back to 18.8.2 but still not able to build for both native and Arduino.  Admittedly, some of this may be user error though

jwellbelove, 2020-11-04:
Ah, is it possible you could list the files?
It's going to be tricky just searching for functions with ternary operators.

rolan-reznik, 2020-11-04:
> Ah, is it possible you could list the files?

Sorry probably I was not clear. I meant that error was not only about `value ? data |= (pattern & MASK) : data &= (~pattern & MASK)` method in flags, but about other places as well. I could try to take a working branch and test.

jwellbelove, 2020-11-04:
Actually, it may be easier just to not use constexpr at all in the ETL if using the Arm5cc compiler?

rolan-reznik, 2020-11-04:
There are places where it works and in my experience `constexpr` gives performance/space benefits. This issue is strange, because armcc5 documentation  states that `constexpr` c++11 is fully supported.

rolan-reznik, 2020-11-04:
Excerpt from https://www.keil.com/support/man/docs/armcc/armcc_chr1407404265784.htm
```
Fully supported C++11 features
ARM Compiler fully supports the following language features as defined by the C++11 language standard:
auto can be used as a type specifier in declaration of a variable or reference.
constexpr.
...
```

rolan-reznik, 2020-11-04:
Do you need help for testing fixes?

lfenster, 2020-11-05:
if it helps, i think this is similar to the issue I was having on this current version.  running with 18.9.2 (instead of 18.8.2) gives me this error output:

In file included from .pio\libdeps\nativetest\Embedded Template Library\include/etl\basic_string.h:55,
                 from .pio\libdeps\nativetest\Embedded Template Library\include/etl\string.h:35,
                 from lib\motorcommand/stringhelper.h:4,
                 from lib\motorcommand/motorcommand.h:6,
                 from test\testmotorcommand\testmotorcommand.cpp:1:
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h: In instantiation of 'constexpr etl::flags<T, MASK>& etl::flags<T, MASK>::set() const [with T pattern = 1; bool value = false; T = unsigned char; T MASK = 255]':
.pio\libdeps\nativetest\Embedded Template Library\include/etl\basic_string.h:253:39:   required from here
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:106:20: error: assignment of member 'etl::flags<unsigned char>::data' in read-only object
  106 |       value ? data |= (pattern & MASK) : data &= (~pattern & MASK);
      |               ~~~~~^~~~~~~~~~~~~~~~~~~
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:106:47: error: assignment of member 'etl::flags<unsigned char>::data' in read-only object
  106 |       value ? data |= (pattern & MASK) : data &= (~pattern & MASK);
      |                                          ~~~~~^~~~~~~~~~~~~~~~~~~~
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:108:15: error: binding reference of type 'etl::flags<unsigned char>&' to 'const etl::flags<unsigned char>' discards qualifiers
  108 |       return *this;
      |               ^~~~
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:109:5: error: body of 'constexpr' function 'constexpr etl::flags<T, MASK>& etl::flags<T, MASK>::set() const [with T pattern = 1; bool value = false; T = unsigned char; T MASK = 255]' not a return-statement
  109 |     }
      |     ^
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h: In instantiation of 'constexpr etl::flags<T, MASK>& etl::flags<T, MASK>::set() const [with T pattern = 2; T = unsigned char; T MASK = 255]':
.pio\libdeps\nativetest\Embedded Template Library\include/etl\basic_string.h:263:35:   required from here
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:124:12: error: assignment of member 'etl::flags<unsigned char>::data' in read-only object
  124 |       data |= (pattern & MASK);
      |       ~~~~~^~~~~~~~~~~~~~~~~~~
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:126:15: error: binding reference of type 'etl::flags<unsigned char>&' to 'const etl::flags<unsigned char>' discards qualifiers
  126 |       return *this;
      |               ^~~~
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:127:5: error: body of 'constexpr' function 'constexpr etl::flags<T, MASK>& etl::flags<T, MASK>::set() const [with T pattern = 2; T = unsigned char; T MASK = 255]' not a return-statement
  127 |     }
      |     ^
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h: In instantiation of 'constexpr etl::flags<T, MASK>& etl::flags<T, MASK>::set(bool) const [with T pattern = 1; T = unsigned char; T MASK 
= 255]':
.pio\libdeps\nativetest\Embedded Template Library\include/etl\basic_string.h:292:38:   required from here
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:115:20: error: assignment of member 'etl::flags<unsigned char>::data' in read-only object
  115 |       value ? data |= (pattern & MASK) : data &= (~pattern & MASK);
      |               ~~~~~^~~~~~~~~~~~~~~~~~~
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:115:47: error: assignment of member 'etl::flags<unsigned char>::data' in read-only object
  115 |       value ? data |= (pattern & MASK) : data &= (~pattern & MASK);
      |                                          ~~~~~^~~~~~~~~~~~~~~~~~~~
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:117:15: error: binding reference of type 'etl::flags<unsigned char>&' to 'const etl::flags<unsigned char>' discards qualifiers
  117 |       return *this;
      |               ^~~~
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:118:5: error: body of 'constexpr' function 'constexpr etl::flags<T, MASK>& etl::flags<T, MASK>::set(bool) const [with T pattern = 1; T = 
unsigned char; T MASK = 255]' not a return-statement
  118 |     }
      |     ^
*** [.pio\build\nativetest\test\testmotorcommand\testmotorcommand.o] Error 1
========================================================================================= [FAILED] Took 9.48 seconds =========================================================================================

Processing testmotorcontroller in nativetest environment
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Building...
In file included from lib\motor\motor.cpp:1:
lib\motor\motor.h:3:10: fatal error: Servo.h: No such file or directory

***************************************************************
* Looking for Servo.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Servo.h"
* Web  > https://platformio.org/lib/search?query=header:Servo.h
*
[... cut ...]
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:117:15: error: binding reference of type 'etl::flags<unsigned char>&' to 'const etl::flags<unsigned char>' discards qualifiers
  117 |       return *this;
      |               ^~~~
.pio\libdeps\nativetest\Embedded Template Library\include/etl\flags.h:118:5: error: body of 'constexpr' function 'constexpr etl::flags<T, MASK>& etl::flags<T, MASK>::set(bool) const [with T pattern = 1; T = 
unsigned char; T MASK = 255]' not a return-statement
  118 |     }
      |     ^

lfenster, 2020-11-05:
i can confirm this is what i am seeing as well and what is reported on the issue re: strings.h; although it also occurs using vector. Error output below if it helps.  Is there a workaround for this?  @jwellbelove, is there a flag to not use constexpr?  Is that what you were suggesting above? 
_In file included from .pio\libdeps\arduino\Embedded Template Library\include/etl\algorithm.h:43:0,
                 from .pio\libdeps\arduino\Embedded Template Library\include/etl\basic_string.h:40,
                 from .pio\libdeps\arduino\Embedded Template Library\include/etl\string.h:35,
                 from lib\motorcommand\stringhelper.h:4,
                 from lib\motorcommand\motorcommand.h:6,
                 from lib\motorcommand\motorcommand.cpp:2:
Compiling .pio\build\arduino\FrameworkArduino\HardwareSerial2.cpp.o
.pio\libdeps\arduino\Embedded Template Library\include/etl\type_traits.h:69:25: fatal error: type_traits: No such file or directory_

jwellbelove, 2020-11-05:
@lfenster That was my suggestion, to define ETL CONSTEXPR as 'nothing' if the arm5cc was detected.

jwellbelove, 2020-11-05:
Your error above is due to the Arduino not supporting parts of 'standard' C++.

jwellbelove, 2020-11-05:
> Do you need help for testing fixes?

Help identifying all of the places where arcm5cc has issues would be good.
It would be interesting to see if there is a pattern to what code is dislikes.

rolan-reznik, 2020-11-07:
Hello,
I have checked hotfix branch. And I got following errors for flags.h

armcc5:
Lines 130, 138, 146, 156, 167, 175, 185, 194, 202, 283, 301, 311, 321, 331, 341 - "statement may not appear in a constexpr function" .

gcc4.8 
All places where data is modified - "Assignement of member etl::flags<unsigned char>::data in readonly object."

I have tried to investigate why I get errors for gcc and found that for c++11 `constexper` methods are also implicitly `const`. That was changed with c++14.

https://en.cppreference.com/w/cpp/language/constexpr  

Standart draft: §7.1.5.8

jwellbelove, 2020-11-07:
Thanks for that investigation, that's a great help

jwellbelove, 2020-11-08:
The issue should be solved if the ETL_CONSTEXPR is replaced with ETL_CONSTEXPR17 in the lines that fail.

jwellbelove, 2020-11-08:
Sorry, make that ETL_CONSTEXPR14

rolan-reznik, 2020-11-08:
I have tried to replace all places where `data` is changed with ETL_CONSTEXPR14 and that worked both for armcc5 and gcc.
So armcc5 seems correctly implementing `constexpr`, but error message could be better. :)

Also I have found that gcc4 with c++11 and stl is used can't compile type_traits.h. It seems that `gcc < 5` have limited support of type traits:
https://gcc.gnu.org/onlinedocs/gcc-4.9.4/libstdc++/manual/manual/status.html#status.iso.2011

```
Missing is_trivially_copyable, is_trivially_constructible, is_trivially_default_constructible, is_trivially_copy_constructible,
is_trivially_move_constructible, is_trivially_assignable, is_trivially_default_assignable, is_trivially_copy_assignable, is_trivially_move_assignable
```

jwellbelove, 2020-11-13:
These should all be fixed in v18.20.0

closed 2020-11-13 by jwellbelove