ETLCPP/etl #298: etl::flat_map C++17 initialization example not working

diggit, 2020-11-04:
Hi,
I am trying to have braced list initialized etl::flat_map as in example:
```c++
etl::flat_map data{ etl::pair{0, 1}, etl::pair{2, 3}, etl::pair{4, 5}, etl::pair{6, 7} };
```
but compilation fails:
```
error: no matching function for call to 'etl::flat_map<int, int, 4, etl::less<int> >::flat_map(<brace-enclosed initializer list>)'

    7 | etl::flat_map<int, int, 4, etl::less<int> >data { etl::pair{0, 1}, etl::pair{2, 3}, etl::pair{4, 5}, etl::pair{6, 7} };
      |                      
```
More in example in [Compiler explorer](https://godbolt.org/z/f4zvbs)

jwellbelove, 2020-11-04:
Are you using the STL?

diggit, 2020-11-04:
STL is not explicitly disabled. In fact, Compiler explorer can handle only one file and thus etl_profile.h is non existent. 

In the end, problem is, that key type is expected to be `const`. So this works...
```c++
using pair = etl::pair<const int, int>;
etl::flat_map data { pair{0, 1}, pair{2, 3}, pair{4, 5}, pair{6, 7} };
```

Is is expected behavior?

jwellbelove, 2020-11-04:
It follows the STL convention.
For `std:map` the `value_type` is `std::pair<const Key, T>`

jwellbelove, 2020-11-04:
OK, I see that the STL does some conversion from `pair<int, int>` to `pair<const int, int>`

jwellbelove, 2020-11-05:
Actually, I think I see what the underlying issue is.

Initializer list construction is only available if you are using the STL (the ETL doesn't have compiler specific implementations for initializer_list yet).

If you are using the STL then containers that use 'pair' will use `std::pair`. This is to maintain compatibility with the STL as many parts assume parameters to be `std::pair`.

This means that `etl::flat_map data{ etl::pair{0, 1}, etl::pair{2, 3}, etl::pair{4, 5}, etl::pair{6, 7} };` is not valid.
There would probably need to be a conversion operator defined for `etl::pair` to `std::pair` for it to work.

jwellbelove, 2020-11-05:
If you want code to be compilable both with and without the STL then you must use `ETL_OR_STD::pair`.

closed 2020-11-13 by jwellbelove