ETLCPP/etl #299: etl::make_string fails for empty strings

twam, 2020-11-05:
With 18.19.2 creating an empty string with etl::make_string like

```
#include <etl/string.h>

int main() {
    const auto emptyString = etl::make_string("");
}
```

fails (see https://godbolt.org/z/WvYhdT).

This was working fine with 17.7.1.

jwellbelove, 2020-11-05:
Yes, the issue has occurred since strings with external buffers were introduced in 18.1.0.
make_string("") tries to create a string of length 0, which now indicates that the string has an external buffer.
I'll push a fix.

jwellbelove, 2020-11-05:
The fix does have the side effect of `max_size()` returning `1` instead of `0`.

jwellbelove, 2020-11-05:
Actually, I have a fix now that has no side effects.

jwellbelove, 2020-11-13:
Fixed in 18.20.0

closed 2020-11-13 by jwellbelove