ETLCPP/etl #303: ETL not compatible for Arduino IDE

glaure, 2020-11-16:
Hi,

This issue happens with Arduino IDE only.
I did try to install the Embedded Template Library using the library manager.
[image]

During installation following messages are written to the output pane:

Sketch uses 10644 bytes (2%) of program storage space. Maximum is 524288 bytes.
Invalid library found in C:\Users\glaure\Documents\Arduino\libraries\Embedded_Template_Library: no headers files (.h) found in C:\Users\glaure\Documents\Arduino\libraries\Embedded_Template_Library
Invalid library found in C:\Users\glaure\Documents\Arduino\libraries\Embedded_Template_Library: no headers files (.h) found in C:\Users\glaure\Documents\Arduino\libraries\Embedded_Template_Library
Invalid library found in C:\Users\glaure\Documents\Arduino\libraries\Embedded_Template_Library: no headers files (.h) found in C:\Users\glaure\Documents\Arduino\libraries\Embedded_Template_Library
Invalid library found in C:\Users\glaure\Documents\Arduino\libraries\Embedded_Template_Library: no headers files (.h) found in C:\Users\glaure\Documents\Arduino\libraries\Embedded_Template_Library

A simple sketch trying to import <etl/vector.h> fails:

sketch_nov16a:1:24: fatal error: etl/vector.h: No such file or directory
 #include <etl/vector.h>
                        ^
compilation terminated.
exit status 1
etl/vector.h: No such file or directory

The library does not follow the specification for an Arduino library and maybe should no longer be provided by the Arduino IDE library manager.

jwellbelove, 2020-11-16:
I have considered dropping support for Arduino, as there are a few things that the ETL expects to find in a conforming C++ compiler that the Arduino lacks.

lfenster, 2020-11-22:
I think this happens in PlatformIO as well.  I dont think its an IDE issue.  I think there is something happening where including both <Arduino.h> and some of the etlcpp ones cause this kind of issue to occur.  still trying to determine why....

bmcdonnell, 2020-12-18:
Looks like my issue #318 (reported later) is a dupe of this, but there's more productive discussion happening there, so maybe this issue could be closed.

jwellbelove, 2020-12-19:
Rather than modify the ETL repository directory layout or require scripts to be run on download or force users to use a ZIP library, I am creating an Arduino ETL repository which will be a fork of the latest ETL version, but with an Arduino compatible folder layout and minus the unit tests.

At the moment I am trying to figure out how I can update the Arduino's user contributed library entry for the ETL. They don't seem to make it very easy to find out.

closed 2020-12-19 by jwellbelove