ETLCPP/etl #304: String copy constructor implicitly removed

ashesman, 2020-11-17:
Sorry if this issue is due to my lack of C++ skills, but I get the following error when compiling with etl.  Compiling the following results in the below errors.  Compiler type and settings can be seen in the debug output.

#include "etl/string.h"
void main(void)
{
}

arm-none-eabi-g++ "../Source/BuildInfo.cpp" -mcpu=cortex-m7 -std=gnu++14 -g3 -DUSE_HAL_DRIVER -DSTM32H743xx -DDEBUG -c -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Inc" -I"D:/10214TestFixturePCBFirmware/ThirdParty" -I"D:/10214TestFixturePCBFirmware/ThirdParty/etl/include" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/STM32H7xx_HAL_Driver/Inc" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/STM32H7xx_HAL_Driver/Inc/Legacy" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/CMSIS/Device/ST/STM32H7xx/Include" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/CMSIS/Include" -O0 -ffunction-sections -fdata-sections -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-use-cxa-atexit -Wall -fstack-usage -MMD -MP -MF"Source/BuildInfo.d" -MT"Source/BuildInfo.o" --specs=nano.specs -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -o "Source/BuildInfo.o"
arm-none-eabi-g++ "../Source/CommandManager.cpp" -mcpu=cortex-m7 -std=gnu++14 -g3 -DUSE_HAL_DRIVER -DSTM32H743xx -DDEBUG -c -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Inc" -I"D:/10214TestFixturePCBFirmware/ThirdParty" -I"D:/10214TestFixturePCBFirmware/ThirdParty/etl/include" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/STM32H7xx_HAL_Driver/Inc" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/STM32H7xx_HAL_Driver/Inc/Legacy" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/CMSIS/Device/ST/STM32H7xx/Include" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/CMSIS/Include" -O0 -ffunction-sections -fdata-sections -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-use-cxa-atexit -Wall -fstack-usage -MMD -MP -MF"Source/CommandManager.d" -MT"Source/CommandManager.o" --specs=nano.specs -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -o "Source/CommandManager.o"
arm-none-eabi-g++ "../Source/main.cpp" -mcpu=cortex-m7 -std=gnu++14 -g3 -DUSE_HAL_DRIVER -DSTM32H743xx -DDEBUG -c -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Inc" -I"D:/10214TestFixturePCBFirmware/ThirdParty" -I"D:/10214TestFixturePCBFirmware/ThirdParty/etl/include" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/STM32H7xx_HAL_Driver/Inc" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/STM32H7xx_HAL_Driver/Inc/Legacy" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/CMSIS/Device/ST/STM32H7xx/Include" -I"D:/10214TestFixturePCBFirmware/STM32CubeMx/Drivers/CMSIS/Include" -O0 -ffunction-sections -fdata-sections -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-use-cxa-atexit -Wall -fstack-usage -MMD -MP -MF"Source/main.d" -MT"Source/main.o" --specs=nano.specs -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -o "Source/main.o"
In file included from ../Source/main.cpp:12:0:
D:/10214TestFixturePCBFirmware/ThirdParty/etl/include/etl/string.h: In static member function 'static etl::private_string_char::make_string_helper<1>::type etl::private_string_char::make_string_helper<1>::make(const char (&)[1])':
D:/10214TestFixturePCBFirmware/ThirdParty/etl/include/etl/string.h:504:44: error: use of deleted function 'etl::string<0>::string(const etl::string<0>&)'
         return etl::string<0U>(text, &c, 1U);
                                            ^
D:/10214TestFixturePCBFirmware/ThirdParty/etl/include/etl/string.h:269:9: note: 'etl::string<0>::string(const etl::string<0>&)' is implicitly deleted because the default definition would be ill-formed:
   class string<0U> : public istring
         ^~~~~~~~~~
D:/10214TestFixturePCBFirmware/ThirdParty/etl/include/etl/string.h:269:9: error: 'etl::ibasic_string<T>::ibasic_string(const etl::ibasic_string<T>&) [with T = char]' is private within this context
In file included from D:/10214TestFixturePCBFirmware/ThirdParty/etl/include/etl/string.h:35:0,
                 from ../Source/main.cpp:12:
D:/10214TestFixturePCBFirmware/ThirdParty/etl/include/etl/basic_string.h:2252:5: note: declared private here
     ibasic_string(const ibasic_string&);
     ^~~~~~~~~~~~~

jwellbelove, 2020-11-17:
Thanks for the heads-up, I'll look into that.

ashesman, 2020-11-18:
This issue is not present in rev 4aed78e (previous release).  Hope that helps to track it down.

jwellbelove, 2020-11-18:
OK, I highlighted the problem by adding an explicitly deleted copy constructor for the string specialisation that uses external buffers.
I'm adding a change that will fix the compilation error, but I think the longer term fix is to make the 'external buffer' containers a separately defined class, rather than a specialisation of the normal container.
I'll let you know when I've uploaded the change.

jwellbelove, 2020-11-18:
If you want to do a temporary fix, then delete the definitions of `make_string` from the ETL headers.

labodj, 2020-11-18:
@jwellbelove Hi!
I'm not sure it's the same problem, but compiling 18.20.0 for ESP32 using Arduino Framework gives me these errors:

```
In file included from src/main.cpp:2:0:
.pio/libdeps/esp32/Embedded Template Library/include/etl/string.h: In static member function 'static etl::private_string_char::make_string_helper<1u>::type etl::private_string_char::make_string_helper<1u>::make(const char (&)[1])':
.pio/libdeps/esp32/Embedded Template Library/include/etl/string.h:499:44: error: use of deleted function 'etl::string<0u>::string(const etl::string<0u>&)'
         return etl::string<0U>(text, &c, 1U);
                                            ^
.pio/libdeps/esp32/Embedded Template Library/include/etl/string.h:266:9: note: 'etl::string<0u>::string(const etl::string<0u>&)' is implicitly deleted because the default definition would be ill-formed:
   class string<0U> : public istring
         ^
In file included from .pio/libdeps/esp32/Embedded Template Library/include/etl/string.h:35:0,
                 from src/main.cpp:2:
.pio/libdeps/esp32/Embedded Template Library/include/etl/basic_string.h:2252:5: error: 'etl::ibasic_string<T>::ibasic_string(const etl::ibasic_string<T>&) [with T = char]' is private
     ibasic_string(const ibasic_string&);
     ^
In file included from src/main.cpp:2:0:
.pio/libdeps/esp32/Embedded Template Library/include/etl/string.h:266:9: error: within this context
   class string<0U> : public istring
         ^
*** [.pio/build/esp32/src/main.cpp.o] Error 1
```

main.cpp:

```cpp
#include <Arduino.h>
#include <etl/string.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

platformio.ini

```ini
[env:esp32]
platform = espressif32
framework = arduino
board = esp32dev

lib_deps = etlcpp/Embedded Template Library @ 18.20.0
```

jwellbelove, 2020-11-18:
Looks like the same issue. I'm not sure why none of the unit tests highlight this!
I use VS2019, GCC 8 & Clang 9 & 10

jwellbelove, 2020-11-18:
I've pushed a fixed in 18.20.1

ashesman, 2020-11-18:
Good stuff.  Thanks for being so responsive...

labodj, 2020-11-18:
I can confirm that 18.20.1 compiles fine! Thanks for your time and dedication @jwellbelove !

closed 2020-11-26 by jwellbelove