ETLCPP/etl #312: Unknown new-header on Arduino megaAVR Boards

HpLightcorner, 2020-12-03:
**Issue**
When compiling the ETL for the Arduino Nano-Every, it fails due to a missing include in `<memory.h>`. For some reason, `avr-gcc` does not handle `<new>`, only `<new.h>`.

**Suggested Solution**
change `#include <new>` to `#include <new.h>` and `#include <memory>` to `#include <memory.h>`. This will for sure not affect any other platform I guess?

I will make a PR - Tests are running fine using CLANG and MSVC and avr-gcc with Arduino Nano-Every.

HpLightcorner, 2020-12-03:
Okay, learned something new [here](https://stackoverflow.com/questions/301586/difference-between-using-includefilename-and-includefilename-h-in-c). 

I understand that `#include <new>` is a good idea to make all features of the header available, so the best solution might be to add another define like `#define ETL_INCLUDE_C_STYLE_NEW_HEADER ` to do the following:

```
#ifdef ETL_INCLUDE_C_STYLE_NEW_HEADER
#include <new.h>
#else
#include <new>
#endif
```

HpLightcorner, 2020-12-03:
Some further information: The Arduino-Core for megaAVR has a custom `new.h` and `new.cpp` file, it looks like the reason why ETL compiles with `#include <new.h>`

I had a look at ETL and `#include <new>` seems to be very common in several files. Isn't that a somehow an unwanted dependency?

jwellbelove, 2020-12-03:
Thanks for the suggestions. I'll take a look at that.
Arduino compatibility has always been an issue for the ETL, as it leaves out parts of the standard library and headers that other platforms provide.

HpLightcorner, 2020-12-04:
@jwellbelove have you seen my PR on that one?

jwellbelove, 2020-12-04:
OK, just seen it. I'll check it out.

jwellbelove, 2020-12-05:
I've been trying this code for Arduino 1.8.13

```
#include <new>

int i;

void setup() {
  // put your setup code here, to run once:
  ::new ((void*)&i) int(1);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
I haven't been able to get it to complain about `#include <new>` yet.

jwellbelove, 2020-12-05:
I'm looking at automatic detection (if not explicitly defined) for the <new> header, in `platform.h`

```
// Figure out if we have a standard library <new> header available.
#if !defined(ETL_USING_STD_NEW)
  #if defined(__has_include)
    #define ETL_USING_STD_NEW __has_include(<new>)
  #elif ETL_USING_STL && !defined(ARDUINO) && !defined(__AVR__)
    #define ETL_USING_STD_NEW 1
  #else
    #define ETL_USING_STD_NEW 0
  #endif
#endif
```

jwellbelove, 2020-12-05:
Ok, I found the board right setup to make it fail. I didn't have the right board support installed.

jwellbelove, 2020-12-06:
I've got the Arduino IDE to successfully compile AVR and non-AVR using this in `platform.h`

```
// Figure out if we can use the standard library <new> header, if haven't already done so in etl_profile.h
#if !defined(ETL_USING_STD_NEW)
  #if defined(__has_include)
    #define ETL_USING_STD_NEW __has_include(<new>)
  #elif ETL_NOT_USING_STL || (defined(ARDUINO) && defined(__AVR__))
    #define ETL_USING_STD_NEW 0
  #else
    #define ETL_USING_STD_NEW 1
  #endif
#endif
```

And `new.h` contains this.

```
#include "platform.h"

#if ETL_USING_STD_NEW
  #include <new>
#else
// Define placement new if no new header is available
inline void* operator new(size_t, void* p) { return p; }
inline void* operator new[](size_t, void* p) { return p; }

inline void operator delete(void*, void* const) { }
inline void operator delete[](void*, void* const) {}
#endif
```

HpLightcorner, 2020-12-06:
@jwellbelove this looks fine, the approach to automatically detect `<new>` also looks valid for me. Did you place the ETL - `new.h` in the private folder like I did in my PR?

jwellbelove, 2020-12-06:
No, I moved it to the 'etl' folder, to allow others to use 'placement new' in a user defined class.

HpLightcorner, 2020-12-06:
That makes sense, but what is going to happen if  framework has defined `new.h` somewhere on the include paths? Not sure if this will end up in some include issues as we only check for `__has_include(<new>)` and not for `__has_include(<new.h>)`? (e.g. the Core Arduino defines that file somewhere, sometimes)

jwellbelove, 2020-12-06:
When calling `#include "new.h"` from the ETL source, it should find the ETL's version first.
User code should be using `#include "etl/new.h"` as the `include/etl` folder should **not** be in the include path list.

Shouldn't `__has_include(<new.h>)` find the Arduino version before the ETL's?

HpLightcorner, 2020-12-06:
Yes, `__has_include(<new.h>)` should find the Arduino header first, but I think in `__has_include(<new>)` won't find `<new.h>` - So I don't think it will find the Arduino header (which by the way has no placement new defined). 

I thin the user-code fits perfect, so I think we can solve this issue in that way.

jwellbelove, 2020-12-06:
I've just had another thought. I think that, if the compiler doesn't define its own `new.h`, `__has_include(<new.h>)` will **always** find the ETL's version.
Maybe the ETL's version should have a unique name?

HpLightcorner, 2020-12-06:
Hmmm, the compiler should only find the ETL `new.h` if the etl folder itself is part of the include paths, which should not be done by the user. So that should not be an issue. But giving it a unique name is also an option!

jwellbelove, 2020-12-06:
I've just checked and you are correct. It doesn't find the ETL's `new.h`.

jwellbelove, 2020-12-06:
I've decided to call the ETL's version `placement_new.h`.

HpLightcorner, 2020-12-08:
@jwellbelove I think we can close this one due to the latest release? (Thanks for your time and effort!)

closed 2020-12-08 by jwellbelove