ETLCPP/etl #314: etl::format_spec not working anymore

HpLightcorner, 2020-12-03:
**Issue**
In a previous version, the following code worked:

```
etl::format_spec ft;
ft.precision(DECIMAL_PRECISION);
etl::to_string(ch.getValue(), buffer, ft, true);
....
```

But with the current version, I get an error:
```
error: assignment of member 'etl::basic_format_spec<etl::ibasic_string<char> >::precision_' in read-only object
```

I have seen that there have been changes to https://github.com/ETLCPP/etl/blob/4bcd734dadcc3f8bc7fddb2528af800f2451c33b/include/etl/basic_format_spec.h#L339

by introducing `ETL_CONSTEXPR` with one of the latest commits.
Is this behaving as expected?

jwellbelove, 2020-12-03:
That isn't what I would expect.
I've tried your code in the unit tests and it compiles without error.

HpLightcorner, 2020-12-03:
@jwellbelove maybe again a `avr-gcc` specific issue?

jwellbelove, 2020-12-03:
What version of GCC are you using?
My unit tests are using v8.1 on Windows + v9.3 on Linux on my machine, and v7.5 / v9.3 in the Github CI.

HpLightcorner, 2020-12-03:
[image]

jwellbelove, 2020-12-03:
I'll try your code in the unit tests on Github CI

HpLightcorner, 2020-12-03:
You can find the code [here](https://github.com/DEWETRON/OXYGEN-SDK-SerialCsv/blob/2d72ce693bbc91c63027b9c3814836d26f60ba2d/firmware/embeddedcsvnode/src/details/Visitors.h#L40).

jwellbelove, 2020-12-03:
Do you get a different result if you do this?

```
etl::format_spec ft = etl::format_spec().precision(DECIMAL_PRECISION);

etl::to_string(ch.getValue(), buffer, ft, true);
```

HpLightcorner, 2020-12-03:
Nope, still the same issue:
```
error: assignment of member 'etl::basic_format_spec<etl::ibasic_string<char> >::precision_' in read-only object
precision_ = static_cast<uint_least8_t>(p);
```

jwellbelove, 2020-12-03:
The Github CI for GCC 7.5 completes without error for this code.

https://github.com/ETLCPP/etl/runs/1492413503?check_suite_focus=true

```
    TEST(test_issue_314)
    {
      etl::string<20> str;

      etl::format_spec ft;
      ft.precision(6);
      etl::to_string(1.23, str, ft, true);
    }
```

HpLightcorner, 2020-12-03:
Also tested now with `arm-none-eabi-g++` and C++11, does not compile. Here the compile-command-flags:

```
arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD
```

jwellbelove, 2020-12-03:
The unit tests compile under C++17
That could be the issue. I may need to make them ETL_CONSTEXPR14.

HpLightcorner, 2020-12-03:
Yes, I think that's it!

jwellbelove, 2020-12-04:
Fixed in 19.3.1
Added parameterised constexpr constructor for C++11

closed 2020-12-04 by jwellbelove