ETLCPP/etl #315: [bit_stream] getting float consumes double bytes

gabbla, 2020-12-03:
Hello, 

while putting float into the `bit_stream` works, getting them consumes 2 times the needed bits (on my platform 64 instead of 32). I'm running this test against latest version, **19.3.0** but I was able to reproduce on older releases too.

Following a snippet where 2 float are pushed into the stream, then extracted:

```c++
#include <etl/array.h>
#include <etl/bit_stream.h>

#include <iostream>

int main(int argc, char *argv[]) {
    constexpr auto SIZE = sizeof(float) * 2;
    std::cout << "sizeof(float): " << sizeof(float) << std::endl;
    char buffer[SIZE];
    etl::bit_stream bs(buffer, SIZE);

    etl::array<float, 2> values{1.2f, 1.3f};

    for(const auto &v : values)
        std::cout << "Put: " << v << " -> " << std::boolalpha << bs.put(v) << std::endl;

    /* Reset */
    bs.restart();
    for(const auto &v : values) {
        float rvalue;
        auto res = bs.get(rvalue);
        std::cout << "Exp: " << v << " -> " << std::boolalpha << res;
        if(res)
            std::cout << " rvalue: " << rvalue;
        std::cout << std::endl;
    }

    return 0;
}
```

Output:

```console
sizeof(float): 4
Put: 1.2 -> true
Put: 1.3 -> true
Exp: 1.2 -> true rvalue: 1.2
Exp: 1.3 -> false
```

I suspect that the bug is located at  317 because `get`, located at line 312, will consume the `bits_remaining` from the stream by itself. 
https://github.com/ETLCPP/etl/blob/4bcd734dadcc3f8bc7fddb2528af800f2451c33b/include/etl/bit_stream.h#L310-L317

Removing line 317 seems to resolve the problem.

gabbla, 2020-12-03:
I'd like to open a pull request, but the company firewall blocks me from pulling/pushing to github from the machine I'm currently working.

jwellbelove, 2020-12-03:
No worries, I'll look at that later today.

jwellbelove, 2020-12-04:
Fixed in 19.3.1

closed 2020-12-04 by jwellbelove