ETLCPP/etl #316: etl::span implementation incomplete

mampfes, 2020-12-03:
The C++ standard defines some overloaded methods for `std::span` which are missing in etl, e.g. the 2nd variant of `first`:

```
template< std::size_t Count >
constexpr std::span<element_type, Count> first() const;
constexpr std::span<element_type, std::dynamic_extent> first( size_type Count ) const;
```

https://en.cppreference.com/w/cpp/container/span/first

Is there a special reason? Would it be ok if I create a PR to add the missing methods?

jwellbelove, 2020-12-03:
Yes, go ahead. Don't forget to add them to the unit tests and check that they pass.

closed 2020-12-09 by mampfes

mampfes, 2020-12-09:
Not sure if I made a mistake here because my PR merged the change into a feature branch of the main repo only. Do I have to create another PR to merge the changes into the master (or development?) branch?

jwellbelove, 2020-12-09:
No the PR was fine. I changed the base to a feature branch. I will check the code in that branch, update the version files and release notes, before merging it to development and master. I usually do changes in a feature or hotfix branch.