ETLCPP/etl #318: Error(?) messages when installing on Arduino IDE

bmcdonnell, 2020-12-14:
When installing Embedded Template Library v19.3.4 from the Library Manager in Arduino IDE 1.8.13, I get many instances of the following messages in the console.

`Invalid library found in C:\Users\me\Documents\Arduino\libraries\Embedded_Template_Library: no headers files (.h) found in C:\Users\me\Documents\Arduino\libraries\Embedded_Template_Library`

I think the relevant info to resolve the issue would probably be found here:
https://arduino.github.io/arduino-cli/latest/library-specification/

bmcdonnell, 2020-12-14:
The error messages repeat when I open the Library Manager. I guess it's iterating through existing library folders to see what's installed.

jwellbelove, 2020-12-15:
Trying to make the ETL compatible with the Arduino IDE and its library manager is a next to impossible task.

The Arduino library system expects the header(s) for the library to be either in the root library folder, or in a sub-directory called 'src'. It adds the library path to its list of include paths.

This has three issues. 

The first is that selecting Sketch / Include Library / Embedded Template Library adds all of the ETL's headers (all 176 of them!).

The second is that it is impossible to select between headers with the same name, as you cannot use a sub-directory name to select the one that you want. To fix this would require me to add a (hopefully) unique prefix to each and every header in the ETL library. 
This is not going to happen.

The third is that everyone, except users of the Arduino IDE, will normally access the library through the standard ETL sub-directory, by setting the additional include directory to `"etl/include"` and use `#include "etl/array.h"`, or similar, in their code.

Due to the Arduino IDE's instance on being completely non-standard and unconfigurable I shall probably have to remove it from the Arduino library manager. 

I shall have to add instructions to the ETL website to show Arduino users how to manually add the library to their projects.

bmcdonnell, 2020-12-15:
How about an export script to reorganize things for Arduino? Or does it
have to pull directly from a GitHub repo? (I guess you could export to
another repo in that case, and have the Arduino library manager use that as
its source, instead of this one.)

If you export the files to `src/etl/`, would we use the `etl/` path, e.g.
`#include <etl/array.h>`?

jwellbelove, 2020-12-15:
_If you export the files to `src/etl/`, would we use the `etl/` path, e.g.
`#include <etl/array.h>`?_

I'm afraid that doesn't work either.

jwellbelove, 2020-12-15:
OK, the Arduino's rules for include are weird!

If I put the ETL headers in `Arduino\libraries\Embedded_Template_Library\src\etl` then `#include <etl/array.h>` doesn't work.

If I put an _empty_ `etl.h` in `Arduino\libraries\Embedded_Template_Library\src` then `#include <etl/array.h>` doesn't work.

If I `#include <etl.h>` before `#include <etl/array.h>` then it does work!!?

The only other option that works is to `#include` _all_ of the headers in `etl.h`.
That may be the only option for Arduino users as `Sketch / Include Library / Embedded Template Library` will add `#include <etl.h>`

They will either have to run a script to move the files, or I will have to create a ZIP library.

HpLightcorner, 2020-12-15:
@jwellbelove 
Have a look at [this](https://github.com/DEWETRON/OXYGEN-SDK-SerialCsv/blob/main/firmware/embeddedcsvnode/create_arduino_library.py) python script. It manages to create a ZIP File where the `<etl/xxx.h>` includes are working as expected. I guess it could be an issue with `etl_profile.h` which have to present at the root of the `src` folder or in the same folder as the `.ino` file? Maybe you can post the verbose ouput when compilation fails. 

In think it is not directly an issue of the Arduino IDE, it simply includes the `src` folder and the root folder of each library to the include paths for the compiler commands.

jwellbelove, 2020-12-15:
The ETL is in `Arduino\libraries\Embedded_Template_Library\src\etl`

`#include <etl/array.h>
`
```
sketch_dec15a:18:10: fatal error: etl/array.h: No such file or directory
 #include <etl/array.h>

```

jwellbelove, 2020-12-15:
With an _empty_ `etl.h`

```
#include <etl.h>
#include <etl/array.h>
```

```
Sketch uses 804 bytes (1%) of program storage space. Maximum is 49152 bytes.
Global variables use 22 bytes (0%) of dynamic memory, leaving 6122 bytes for local variables. Maximum is 6144 bytes.
```

jwellbelove, 2020-12-15:
Which directory do you run the Python in?

All I get is this

```
  File ".\create_arduino_library.py", line 20, in <module>
    shutil.copytree(src, os.path.join(dest, 'src'),  dirs_exist_ok=True)
TypeError: copytree() got an unexpected keyword argument 'dirs_exist_ok'
```

jwellbelove, 2020-12-15:
OK, I have 3.7.3. Looks like I need to update.

jwellbelove, 2020-12-15:
If I run it from the ETL root directory I get this.

```
PS D:\Users\John\Documents\Programming\GitHub\etl> python .\create_arduino_library.py
The destination folder is  D:\Users\John\Documents\Programming\GitHub\etl\embedded-template-library
Traceback (most recent call last):
  File "D:\Users\John\Documents\Programming\GitHub\etl\create_arduino_library.py", line 20, in <module>
    shutil.copytree(src, os.path.join(dest, 'src'),  dirs_exist_ok=True)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\shutil.py", line 555, in copytree
    with os.scandir(src) as itr:
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'D:\\Users\\John\\Documents\\Programming\\GitHub\\etl\\etl/include'
```

jwellbelove, 2020-12-15:
If I run it from the Github directory...

```
PS D:\Users\John\Documents\Programming\GitHub> python .\create_arduino_library.py
The destination folder is  D:\Users\John\Documents\Programming\GitHub\embedded-template-library
Done adding ETL...
Traceback (most recent call last):
  File "D:\Users\John\Documents\Programming\GitHub\create_arduino_library.py", line 26, in <module>
    shutil.copytree(src, os.path.join(dest, 'src'), dirs_exist_ok=True)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\shutil.py", line 555, in copytree
    with os.scandir(src) as itr:
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'D:\\Users\\John\\Documents\\Programming\\GitHub\\src'
```

jwellbelove, 2020-12-15:
I tried changing `src = os.path.join(dirname, 'external/etl/include')` to `src = os.path.join(dirname, 'etl/include')`

jwellbelove, 2020-12-15:
I managed to get it working by editing the python code to this...

```
import shutil
import os

# https://arduino.github.io/arduino-cli/latest/library-specification/

#
# Some Constants
library_name = 'embedded-template-library'

#
# Get the current Path of the script
dirname = os.path.dirname(__file__)
dest = os.path.join(dirname, library_name)
print('The destination folder is ', dest)

#
# Copy the ETL to the Arduino-Library Folder
src = os.path.join(dirname, 'include')
shutil.copytree(src, os.path.join(dest, 'src'),  dirs_exist_ok=True)
print('Done adding ETL...')

#
# Add files from the src folder
#src = os.path.join(dirname, 'src')
#shutil.copytree(src, os.path.join(dest, 'src'), dirs_exist_ok=True)
#print('Done adding Source...')

#
# Add any additional file (library.properties)
filename = 'library.properties'
shutil.copyfile(os.path.join(dirname, filename), os.path.join(dest, filename))
print('Done adding additional files...')

#
# Zip the Folder to be Arduino-Compatible
print('Make archive...')
shutil.make_archive(os.path.join(dirname, library_name), 'zip', dest)

#
# Remove temporary directory
print('Removing temporary files...')
shutil.rmtree(dest)
```

HpLightcorner, 2020-12-15:
> The ETL is in `Arduino\libraries\Embedded_Template_Library\src\etl`
> 
> `#include <etl/array.h> `
> 
> ```
> sketch_dec15a:18:10: fatal error: etl/array.h: No such file or directory
>  #include <etl/array.h>
> ```

Can you turn on verbose output when compiling? (In the Arduino Preferences)

HpLightcorner, 2020-12-15:
> I managed to get it working by editing the python code to this...
>
> ```
> import shutil
> import os
>
> # https://arduino.github.io/arduino-cli/latest/library-specification/
>
> #
> # Some Constants
> library_name [...]

Yeah, the file is made for the firmware of the project I linked which simply includes the ETL source as an additional folder, so there are for sure changes necessary to get it work with the ETL Git folder as it is.

jwellbelove, 2020-12-15:
```
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Users\John\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Users\John\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Users\John\Documents\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=off -ide-version=10813 -build-path S:\Temp\John\arduino_build_761600 -warnings=none -build-cache S:\Temp\John\arduino_cache_152267 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose D:\Users\John\Documents\Arduino\sketch_dec15a\sketch_dec15a.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\Users\John\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\Users\John\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\Users\John\Documents\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=off -ide-version=10813 -build-path S:\Temp\John\arduino_build_761600 -warnings=none -build-cache S:\Temp\John\arduino_cache_152267 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=D:\Users\John\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose D:\Users\John\Documents\Arduino\sketch_dec15a\sketch_dec15a.ino
Using board 'nona4809' from platform in folder: D:\Users\John\Documents\ArduinoData\packages\arduino\hardware\megaavr\1.8.6
Using core 'arduino' from platform in folder: D:\Users\John\Documents\ArduinoData\packages\arduino\hardware\megaavr\1.8.6
Detecting libraries used...
"D:\\Users\\John\\Documents\\ArduinoData\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-ID:\\Users\\John\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-ID:\\Users\\John\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-ID:\\Users\\John\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\nona4809" "S:\\Temp\\John\\arduino_build_761600\\sketch\\sketch_dec15a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for etl/array.h: []
ResolveLibrary(etl/array.h)error: etl/array.h: No such file or directory

  -> candidates: []y.h>

```

jwellbelove, 2020-12-15:
That last output was for `#include <etl/array.h>` on its own.

HpLightcorner, 2020-12-15:
Seems that the IDE fails during the library discovery phase - it only looks if your include file is present either in the `root` folder of the lib or in `root/src`. **If and only if** it can find the include file, it will add the folder to the compile commands. Hence, it fails with the additional subfolder `etl/array.h`. So, when you do `#include <etl.h>`, which needs to be present on the src-folder level, the IDE is able to discover the library and will then add the folder to the compile-commands - which then allows you to stick with the `etl/xxx.h`syntax.

I would suggest to implement the workaround you suggested, the Arduino users need to include `etl.h`, which is simply an empty file but necessary to allow the IDE to find the ETL-Library folder. Also, add the following property to the `properties.json` file: `includes=etl.h`

```
Arduino/
    libraries/
        embedded-template-libray/
            library.properties
            src/
                etl_profile.h
                etl.h
                etl/
                    .....
```

HpLightcorner, 2020-12-15:
The problem never arises during development of [OXYGEN-SDK-SerialCsv](https://github.com/DEWETRON/OXYGEN-SDK-SerialCsv) because the `SerialCsv.h` file is normally included by the user before thinking of any additional ETL-file, hence the Arduino IDE was always able to find the library folder and set the include paths for the compile commands.

jwellbelove, 2020-12-17:
I got the Arduino compiling OK on my machine, but I'm having a weird error when I try to use PlatformIO to compile the same code.

There are two header files that push and pop `min` & `max` macros.
The PlatformIO compilation errors on the line `#pragma pop_macro("max")` with `minmax_pop.h:38:7: error: expected unqualified-id before '(' token`

It seems to be happy with `#pragma pop_macro("min")`, even if I reverse the order of the lines.

Have you used PlatformIO, and if so, come across this at all?

flocked, 2020-12-18:
First thanks for the library.
I unfortunately also can't get the library to install in Arduino. I tried the edit python script. it creates a zip. But trying to import it, Arduino tells me, it's an invalid zip, not containing any library.
Do I maybe have the folder structure wrong?

jwellbelove, 2020-12-18:
I'm working on the script myself at the moment. The key for the Arduino is the have a `.h` file in the `src` directory which you include. It doesn't matter if this header is empty. Once the Arduino has seen the header, then it is happy that it has a valid library.

jwellbelove, 2020-12-18:
There is the option that a python/windows/linux script could be made available for the user to run to rename the `include` directory to `src`.

Other options would be to make a breaking change by  rename the `include` directory to `src` on Github, or create a copy of the code in a `src` directory (not my favourite option)

HpLightcorner, 2020-12-18:
@jwellbelove Why not creating a ZIP-Folder for each release in Github-CI? So the user can simply install ETL by adding a zipped Arduino Library?

I will do a test with platformio, in the [etl_profile.h](https://github.com/DEWETRON/OXYGEN-SDK-SerialCsv/blob/main/firmware/embeddedcsvnode/src/etl_profile.h) I use for Arduino I undef `min` and `max`.

jwellbelove, 2020-12-18:
I've tried that and it works OK, but the library is still available through the Arduino IDE library manager. If you use this route then the library is not valid. The only way around this, if the `include` directory is not renamed in some way, is to request that the ETL is removed from the contributed library list.

jwellbelove, 2020-12-18:
Yet another option is to create a mirrored ETL-Arduino repository on Github and link the Arduino contributed library to this.

flocked, 2020-12-18:
I now also got it working in Arduino. I also tested it with PlatformIO and my Teensy4.1. It works great. 

I would however strongly suggest to add a note somewhere that library headers need to include` <etl/… ` or update the ETL examples inside PlatformIO. I tried it without like in all examples for half an hour and thought the library would be broken. :D

jwellbelove, 2020-12-19:
Strangely, I'm not getting the `#pragma pop_macro("max")` error today. No idea what changed!

jwellbelove, 2020-12-19:
Rather than modify the ETL repository directory layout or require scripts to be run on download or force users to use a ZIP library, I am creating an Arduino ETL repository which will be a fork of the latest ETL version, but with an Arduino compatible folder layout and minus the unit tests.

At the moment I am trying to figure out how I can update the Arduino's user contributed library entry for the ETL. They don't seem to make it very easy to find out.

bmcdonnell, 2020-12-19:
> Rather than modify the ETL repository directory layout or require scripts to be run on download or force users to use a ZIP library, I am creating an Arduino ETL repository which will be a fork of the latest ETL version, but with an Arduino compatible folder layout and minus the unit tests.

I figure you'd want to create a script to do the "exporting" (selective copying to the new layout), and then capture that export in the new repo. It wouldn't need to be a fork, except for being able to install old (broken on Arduino?) versions.

> At the moment I am trying to figure out how I can update the Arduino's user contributed library entry for the ETL. They don't seem to make it very easy to find out.

Maybe similar to how you submitted it in the first place?

jwellbelove, 2020-12-19:
When I said 'fork' I actually meant 'copy to the other repo's directory on my machine, commit & push'.

I'm pretty sure that I added it by editing the library list on the 'playground'. This doesn't seem to be editable anymore, unless I'm looking in the wrong place.

bmcdonnell, 2020-12-19:
https://support.arduino.cc/hc/en-us/articles/360012175419-How-can-I-add-my-library-to-the-Library-Manager-

closed 2021-01-08 by jwellbelove