ETLCPP/etl #321: Arduino searchability

bmcdonnell, 2020-12-18:
It took me a while to find the library in the Arduino Library Manager. It would help if the text "ETL" were included somewhere that the search field looks at. Apparently both the `name` and `sentence` fields in the `library.properties` file are options. (There may be others, IDK.)

Not found by searching "ETL":

[image]

Found by searching "embedded template":

[image]

Searching by a word in the `sentence` field:

[image]

jwellbelove, 2020-12-19:
Would this work better?

name=Embedded Template Library (ETL)
version=19.3.5
author= John Wellbelove <john.wellbelove@etlcpp.com>
maintainer=John Wellbelove <john.wellbelove@etlcpp.com>
license=MIT
sentence=ETL. A C++ template library tailored for embedded systems.
paragraph=
category=Other
url=https://www.etlcpp.com/
architectures=*

bmcdonnell, 2020-12-19:
Yup!

>

jwellbelove, 2020-12-19:
Rather than modify the ETL repository directory layout or require scripts to be run on download or force users to use a ZIP library, I am creating an Arduino ETL repository which will be a fork of the latest ETL version, but with an Arduino compatible folder layout and minus the unit tests.

At the moment I am trying to figure out how I can update the Arduino's user contributed library entry for the ETL. They don't seem to make it very easy to find out.

closed 2021-01-08 by jwellbelove