ETLCPP/etl #322: The right collection type

flocked, 2020-12-19:
First thanks for the great library.

i really hope it’s okay to ask this question here. I need some advice for using this library.

My project contains several objects where structs (of one type) can be dynamically added or removed. The whole project can only save 100 of these stucts. There isn’t enough ram.

My first thought was to create a pool of 1000. But then I’m not sure how to save structs to objects. With an array? Wouldn’t I need to create the arrays with a size of 1000 for each object, just in case all values get assaigned to only one object? I don’t have memory for that.

Second thought was using the linked_list and it’s pool function.  But if I understand it correctly it will also create quite a lot of additional memory usage as it adds a forward and backward pointer to each value. Adding these + 10000 would tripple the memory usage. 

Any idea? Thanks.

jwellbelove, 2020-12-24:
Hi, sorry for the delay in replying.

From what I'm understanding from your description, you have N objects (10?) and up to 1000 structs. Each object may contain 0 to 1000 of these structs, though no more than 1000 in total.
Is that correct?

One way of doing this would be to use an `etl:: forward_list_ext`. This is a `forward_list` that uses an external pool. Each object would contain one of these forward lists, but they would all refer to the same shared pool. That way you never allocate space for more than 1000 structs, but each object may store any number of them, as long as the total for all objects is not greater than the pool size.

flocked, 2021-01-06:
Sorry for my late reply. Yes that's exactly how I want it.

closed 2021-01-06 by flocked