ETLCPP/etl #323: Support an owning version of delegate

humanosc, 2021-01-08:
I think the etl would greatly benefit of a owning version of the delegate class similar to std::function but instead it stores the closure in an internal buffer (being it is on the stack with a user-defined static capacity) instead of heap allocated memory. The internal buffer size could be spezified by a template parameter. 

Some implementations I found are:
- https://github.com/rigtorp/Function
- https://github.com/tacticalmelonfarmer/callable

humanosc, 2021-01-09:
I have already tried both implementations and recognized that the callable project needs c++17. There are some decent dependencies to the std lib like `std::is_function_v`. It seems there is some effort needed to get this code working based on c++11 or c++14.

In this matter, I also tried The `Function` project which works quite good. I was able to store closures in this Function object which allowed me to call them even when the initial lambda is gone out of scope. This code seems also very compact and genious. At this moment my favourite for the etl.

jwellbelove, 2021-01-10:
Function looks interesting. I quick scan of the file shows that it is defined in a very similar way to etl::delegate.

daniel-brosche, 2021-09-28:
I have adapted some small parts for regarding copy construction https://github.com/daniel-brosche/Function/tree/bugfix/copy_construction_fails

This was needed to achieve something like that:
```cpp

 struct Data {
        Function<void()> on_constructor;
        Function<void()> on_destructor;

        Data(const Function<void()>& on_constructor, const Function<void()>& on_destructor) {
            this->on_constructor = on_constructor;
            this->on_destructor = on_destructor;

            this->on_constructor();
        }

        ~Data() { this->on_destructor(); }
    };

 void test() {
    bool on_constructor_called = false;
    bool on_destructor_called = false;

    Function<void()> on_constructor = [&]() { on_constructor_called = true; };
    Function<void()> on_destructor = [&]() { on_destructor_called = true; };

    Data* d = new Data(on_constructor, on_destructor);
    delete d;
 }

```

MrVichr, 2021-10-10:
Any progress on this? I tried to make some changes and they seem to work, but are somewhat fishy.
- lambdas need to be memmove()'d
- the invocation_element's storage size needs to be increased
- all stubs need to take &invocation.object rather than invocation.object

I can try to make a pull request but 1) I see you have uncommited changes, 2) I'm not sure how to do that so might screw up something.

humanosc, 2021-10-10:
At First, nice to read that you support this feature. 

> * the invocation_element's storage size needs to be increased
 
Why should it be increased? The Idea is to define the max. capacity which aligns with the etl philosophy.
If the capacity/size of the buffer is too small then throw an error/assertion/exception.

MrVichr, 2021-10-11:
The official version works as long as the lambda does not go out of scope. The following test fails with access violation, usually at dele(11); Ideally, it should pass the last assert.
TestDelegate.hpp: https://pastebin.com/ZyuiS0Cm
TestDelegate.cpp: https://pastebin.com/i2mG0zRH

daniel-brosche's version starts with #include <functional>, I don't like that.

@jwellbelove thank you, I'll submit it in about 12 hours from now.

daniel-brosche, 2021-10-11:
I have studied the code of the `Function ` class for a while. I think this implementation is pretty straight forward in comparison to other implementations (like std::function). There are a couple of similiarities to the `etl::delegate`. It is possible to replace some parts with non std helper like `std::aligned_storage` but not sure if it's worth or helpful for the etl to do that.

daniel-brosche, 2021-10-12:
In Function class std::aligned_storage is used. Your code looks pretty similar to the Function class except some memory operations.

daniel-brosche, 2021-10-13:
> An additional `etl::owning_delegate` (just the first name that came to mind), that more closely resembles Function, may be a better option.

I have read in a proposal about a std::function_view as non owning version of std::function analogous to std::string_view / std::string. Therefore `delegate_view` and `delegate` seems for me also appealing but this will result in a breaking change.

daniel-brosche, 2021-10-15:
Did you push this experiment somewhere?

jwellbelove, 2021-10-16:
An owning delegate must implement a destructor that destructs the copy of the lambda. That means the delegate can not be `constexpr`.

i.e. If a destructor is defined for `etl::delegate` then the following code will not compile.

```
static void Func() {}

constexpr etl::delegate<void(void)> d = etl::delegate<void(void)>::create<Func>();
```

MrVichr, 2021-10-18:
I gave it some more effort, and ended up in a template hell, as expected.
It seems it's either destructor, or constexpr. Can't have both. Also delegates containing lambdas can't be constexpr (for now?).

To support correct copying of lambda's captures, I had to introduce a "virtual method table". Now, the invocation_element is *almost* a lambda, in that both support copying, destructing and invoking, but I can't see a way to unite the concepts.

I feel like reinventing the wheel with memory_helper<>. I don't know std:: or etl:: well enough to find something similar.

Further, I'm not having any problems with __GNUC__==8, so I changed the 2 lines that test it.

However ugly, current version seems to pass all my tests. I'd be grateful if anyone can run their test suite against my code. It's accessible at
https://github.com/MrVichr/etl/blob/development/include/etl/delegate.h
More work needs to be done when assigning or comparing incompatible delegates, such as etl::delegate<void(int)> and etl::delegate<int(void)> .

daniel-brosche, 2021-11-09:
I did some small quick performance tests, like that:

```cpp
    bool global_called = false;
    size_t called_id = 0;

    void global(size_t id)
    {
       global_called = !global_called;
       called_id = id;
    }

    const int REP = 10;
    const int IT = 100000000;

    void CallConstExpr() {
        constexpr etl::delegate<void(size_t)> global_callback = etl::delegate<void(size_t)>::create<global>();

        double avg = 0.0;
        for(int k=0; k < REP; k++){
            helper::Timer timer;
            
            for(int i=0; i < IT; i++) {
                global_callback(i);
            }
        
            std::cout<<"constexpr delegate: "<<timer.Elapsed()<<std::endl;
            avg += timer.Elapsed();
        }
        std::cout<<"constexpr delegate avg: "<< std::setprecision(15) << std::fixed << avg / REP<<std::endl;

    }

    void CallConst() {
        etl::delegate<void(size_t)> global_callback = etl::delegate<void(size_t)>::create<global>();
        
        double avg = 0.0;
        for(int k=0; k < REP; k++){
            helper::Timer timer;
            
            for(int i=0; i < IT; i++) {
                global_callback(i);
            }
        
            std::cout<<"delegate: "<<timer.Elapsed()<<std::endl;
            avg += timer.Elapsed();
        }
        std::cout<<"delegate avg: "<< std::setprecision(15) << std::fixed << avg / REP<<std::endl;
    }

```

Here are my results based on GCC 6.3.0 x86_64 compiled with O3:

System A:
CallConstExpr avg: 0.438233127200000
CallConst avg: 0.446093623700000

System B:
CallConstExpr avg: 0.147686385900000
CallConst avg: 0.184802201300000

So, as usual -> it depends...

jwellbelove, 2022-01-23:
Closing due to issues with owning delegates and constexpr.

closed 2022-01-23 by jwellbelove

humanosc, 2022-01-24:
I think to unify both aspects into one Implementation ist not ideal without compromise. But I'm convinced that two seperaten Implementations are quite helpful.