ETLCPP/etl #325: downloadable documentation for offline use?

jerabaul29, 2021-01-23:
It is useful to have a fully downloadable documentation for offline use.

Is there a way to get a full copy of the documentation at https://www.etlcpp.com/ ? Of course I can do a ```wget -r``` of some sort, but that feels a bit rude - is there a better solution? :)

If this is the wrong place to ask sorry, feel free to move this :) .

jwellbelove, 2021-01-23:
I'm afraid there is currently no offline documentation available.
If wget works for you, then do it, though you may miss updates and corrections to the documentation.
Be aware that there are over 160 individual pages on the site!

kassane, 2021-01-23:
@jerabaul29 and @jwellbelove 
The documentation generated by doxygen does not help?
Although I could add more information, because it still seems very raw to me.

I had even contributed Kapeli for ETL docset here:
[#3137](https://github.com/Kapeli/Dash-User-Contributions/pull/3137)

jwellbelove, 2021-01-23:
To be honest, I haven't tried generating the Doxygen documentation for some time.

jerabaul29, 2021-01-23:
To me it looks like the website is the true detailed doc, right? :) I get the 'doxygen content' automatically through my linter anyways, so I was really thinking about the website.

What I was thinking is also that the documentation is / could / maybe should be a 'first class citizen' of the project and, therefore, I was thinking that it may be a good idea to have the sources for the doc living in its own GitHub repo. This would make it very easy to pull it for offline use, update it, comment, open issues, push improvements etc and involve the community in improving the doc. I have met several projects with this organisation - 1 repo for the code, 1 repo for the doc, under the same owner- and it feels like it is a very successful combination and an effective way to leverage the community.

jwellbelove, 2021-01-23:
The web documentation is not plain HTML/CSS. I create it using a WYSIWYG web designer app.

GatorQue, 2021-01-23:
I'm curious, which WYSIWYG web designer app do you use? I have always been impressed with how clean your website looks for the project.

jerabaul29, 2021-01-23:
Mmh, ok, so that means that likely wget is the simplest / best way to pull the documentation I suppose :) .

jwellbelove, 2021-01-23:
I use WYSIWYG Web Builder [https://www.wysiwygwebbuilder.com/](https://www.wysiwygwebbuilder.com/)
I've had a few issues getting the formatting consistent sometimes, but it has a lot of ready made widgets to use on your web site.

I did take a look at Xara Web Designer as I use Xara Photo & Graphic Designer for diagrams and DTP type stuff, but it doesn't have a decent site manager, making large web site designs a bit awkward to navigate.

closed 2021-01-28 by jwellbelove