ETLCPP/etl #326: recommended way to switch between "debug" and "release" modes for bounds-checking

jerabaul29, 2021-01-25:
Let us say I want to switch between some "debug" (full bounds-checkings for arrays, vectors, etc) and "release" (no bounds checking for maximum speed) modes.

What is the recommended way / how would you do so?

Looking at the documentation for vector, available at: https://www.etlcpp.com/vector.html

```
T& at(size_t i)
const T& at(size_t i) const

Returns a reference or const reference to the indexed element. Emits an etl::vector_out_of_range if the index is
out of range of the array. If asserts or exceptions are not enabled then undefined behaviour occurs.
```

Does that mean that a way to achieve this would be to 1) use ```at()``` instead of ```[]``` everywhere, and 2) enable asserts or exceptions for debug mode, and switch them off for release mode? How would you recommend having a global exceptions / asserts switch in the code \?

jwellbelove, 2021-01-25:
The quickest way to disable all checks is to define `ETL_NO_CHECKS`. This stops `ETL_ASSERT` from having any effect at all.

jerabaul29, 2021-01-25:
Ok, so you would recommend using ```.at()``` everywhere, and then putting or not the ```#define ETL_NO_CHECKS  // disables checks for release mode```,  if I understand well? :)

jwellbelove, 2021-01-25:
Yes, you can do that too.

closed 2021-01-28 by jwellbelove