ETLCPP/etl #327: Feature request: getting string format in C++ 20 style?

jerabaul29, 2021-01-25:
C++ 20 received new ways to format strings, in a python-like way (https://en.cppreference.com/w/cpp/utility/format/format), for example:

```
std::format("{} {}!", "Hello", "world", "something"); // OK, produces "Hello world!"
```

This looks amazing, as this is both safe, simple to read and write, and starts to be nearly a "de facto standard" across languages (I finally do not need to get confused when switching back and forth between python and c++, hooray!).

I think that the etl provides something quite similar thanks to the ```to_string``` API (https://www.etlcpp.com/to_string.html). However, would it be possible to get something that is fully similar / compliant to the C++ 20 std::format? That would make it even easier to use, would reduce brain overhead switching between etl and "standard" C++ 20.

jwellbelove, 2021-01-25:
I'll certainly put it on my ToDo list (which is a bit long at the moment!)

jerabaul29, 2021-01-25:
That sounds amazing :) No worries no hurries, and many thanks :) .

evantill, 2022-10-17:
https://github.com/fmtlib/fmt seems to be great and well tested but use _"heavily"_ std library.

jwellbelove, 2022-10-17:
I looked at the code and the use of the STL is very extensive.
I don't think a modification of the code would be very simple or easy.