ETLCPP/etl #329: etl::vector of type Int won't compile with initial_size and initial value constructor (when initial_size is also of type int)

HpLightcorner, 2021-01-28:
Hi,

It seems that I have found I quite "special" bug in etl::vector.

**Toolchain**
Nordic NRF toolchain on Platformio (ARM-GNU)

**Issue**
The following code won't compile:
```
etl::vector<int, 10> test(5, 5);
```
which should call the following constructor:
```
    vector(size_t initial_size, typename etl::ivector<T>::parameter_t value)
      : etl::ivector<T>(reinterpret_cast<T*>(&buffer), MAX_SIZE)
    {
      this->initialise();
      this->resize(initial_size, value);
    }
```
However, compiling fails with the following error:
```
In file included from lib/etl/include/etl/container.h:37,
                 from lib/etl/include/etl/algorithm.h:44,
                 from lib/etl/include/etl/vector.h:41,
                 from src/test.cpp:1:
lib/etl/include/etl/iterator.h: In instantiation of 'struct etl::iterator_traits<int>':
lib/etl/include/etl/vector.h:366:7:   required from 'void etl::ivector<T>::assign(TIterator, TIterator) [with TIterator = int; T = int]'
lib/etl/include/etl/vector.h:1185:7:   required from 'etl::vector<T, MAX_SIZE_>::vector(TIterator, TIterator) [with TIterator = int; T = int; unsigned int MAX_SIZE_ = 10]'
src/test.cpp:10:35:   required from here
lib/etl/include/etl/iterator.h:60:51: error: 'int' is not a class, struct, or union type
     typedef typename TIterator::iterator_category iterator_category;
```

Which indicates that the templated constructor is called instead:
```
    //*************************************************************************
    /// Constructor, from an iterator range.
    ///\tparam TIterator The iterator type.
    ///\param first The iterator to the first element.
    ///\param last  The iterator to the last element + 1.
    //*************************************************************************
    template <typename TIterator>
    vector(TIterator first, TIterator last)
      : etl::ivector<T>(reinterpret_cast<T*>(&buffer), MAX_SIZE)
    {
      this->assign(first, last);
    }
```

**Reason**
This special case arises when the compiler assumes that both parameters passed to the constructor are of the same type. Hence, the templated constructor is called - as this constructor has both parameters to be of template-type. At least this can happen for a vector containing integers and passing `initial_size` of type `int` as well - for sure a special case.

**Solution**
I would suggest adding another constructor explicitly assuming the `initial_size` parameter to be of type `int`:
```
    //*************************************************************************
    /// Constructor, from initial size and value - assuming int as inital_size type
    ///\param initial_size  The initial size of the vector.
    ///\param value        The value to fill the vector with.
    //*************************************************************************
    vector(int initial_size, typename etl::ivector<T>::parameter_t value)
      : etl::ivector<T>(reinterpret_cast<T*>(&buffer), MAX_SIZE)
    {
      this->initialise();
      this->resize(initial_size, value);
    }
```
So that this special case is covered by another constructor telling the compiler not to use the templated one.

If my solution is acceptable I can open a PR.

jwellbelove, 2021-01-28:
I assume std::vector doesn't have this problem. If so I suspect it is using std::enable_if in some way.
All container sizes are unsigned, which would require a cast in the constructor, which I don't want to do.
What it probably needs is a check to disable the iterator constructor for integral types.

jwellbelove, 2021-01-28:
The correct solution is to add this to the iterator constructor

`typename TEnable = etl::enable_if<!etl::is_integral<TIterator>::value, void>::type>`

i.e.
```
template <typename TIterator, typename TEnable = etl::enable_if<!etl::is_integral<TIterator>::value, void>::type>
vector(TIterator first, TIterator last)
  : etl::ivector<T>(reinterpret_cast<T*>(&buffer), MAX_SIZE)
{
  this->assign(first, last);
}
```

I shall check the other containers to see which ones also exhibit this behaviour.

HpLightcorner, 2021-01-29:
Yes, you are completely right, that should be part of the iterator constructor!

jwellbelove, 2021-01-29:
Fixed 19.5.1

closed 2021-01-29 by jwellbelove