ETLCPP/etl #332: etl::string fails to compile without a heap

mkalibat, 2021-02-02:
It's observed that the only difference between failing (linker error) and not-failing builds is an instantiation of etl::string<64>. The linker failure output is:

`/opt/microchip/xc32/v2.49/bin/bin/../../lib/gcc/pic32c/6.2.1/../../../../bin/bin/pic32c-ld Error: A heap is required, but has not been specified`

Does etl::string<> have any methods that dynamically allocate?

Thanks.

jwellbelove, 2021-02-02:
Nothing in the ETL should be allocating from the heap, but I'll investigate anyway.

mkalibat, 2021-02-02:
Thanks!

Note that this is only observed in the debug build (with `DEBUG` macro defined). I see this enables the ETL_DEBUG macro. Are there some print functions or such enabled in debug mode?

jwellbelove, 2021-02-02:
What ETL error handling option macros have you set?

jwellbelove, 2021-02-02:
Could `assert()` be using heap memory?

closed 2021-04-14 by jwellbelove