ETLCPP/etl #334: possible problem using string with external buffer?

jerabaul29, 2021-02-15:
I have some problems using the string with external buffer as described in https://www.etlcpp.com/string.html. I need to have a "usual" cstring as input to other functions, but I would like to work with it as an etl::string for safety otherwise. However, I have some problems. Namely, it seems like the ```etl::string``` is created with size 0, not the size of the buffer. Any idea what I do wrong, or if there is a bug?

```cpp
#include "etl.h"
#include "etl/string.h"

template <size_t size>
void print_etl_string(etl::string<size> etl_string){
  Serial.println(F("--- etl string printout ---"));
  Serial.print(F("print etl string, max size "));
  Serial.print(etl_string.max_size());
  Serial.print(F(" crrt size: "));
  Serial.println(etl_string.size());

  Serial.print(F("content currently in use    : "));
  for (int i=0; i<etl_string.size(); i++){
    Serial.print("0x");
    Serial.print(etl_string[i], HEX);
    Serial.print(F(" | "));
  }
  Serial.println();

  Serial.print(F("content currently not in use: "));
  for (int i=etl_string.size(); i<etl_string.max_size(); i++){
    Serial.print("0x");
    Serial.print(etl_string[i], HEX);
    Serial.print(F(" | "));
  }
  Serial.println();
}

constexpr size_t string_max_size = 8;

// this works fine
// etl::string<string_max_size> etl_string;
/*
 * prints something like:
--- booted ---
0x68 | 0x65 | 0x6C | 0x6C | 0x6F | 
--- etl string printout ---
print etl string, max size 8 crrt size: 0
content currently in use    : 
content currently not in use: 0x0 | 0xEB | 0x0 | 0x0 | 0x94 | 0x72 | 0x0 | 0x0 | 
--- etl string printout ---
print etl string, max size 8 crrt size: 1
content currently in use    : 0x0 | 
content currently not in use: 0x0 | 0x0 | 0x0 | 0x94 | 0x72 | 0x0 | 0x0 | 
 */

// this does not work
// /*
char char_buffer[string_max_size + 1];
etl::string<0> etl_string(char_buffer, etl::size(char_buffer));
#define PRINT_CHAR_BUFF
// */
/*
 * prints
--- booted ---
cstring as hex: 0x68 | 0x65 | 0x6C | 0x6C | 0x6F | 
--- etl string printout ---
print etl string, max size 0 crrt size: 0
content currently in use    : 
content currently not in use: 
--- etl string printout ---
print etl string, max size 0 crrt size: 0
content currently in use    : 
content currently not in use: 
cstring as hex: 0x0 | 
 */

void print_cstring_as_hex(char * start){
  char * crrt_char = start;
  Serial.print(F("cstring as hex: "));

  while (true){
    Serial.print(F("0x"));
    Serial.print(*crrt_char, HEX);
    Serial.print(" | ");
    if (* crrt_char == '\0'){
      break;
    }
    crrt_char ++;
  }
  Serial.println();
}

void setup() {
  Serial.begin(115200);
  Serial.println(F("--- booted ---"));

  print_cstring_as_hex("hello");
}

void loop() {
  print_etl_string(etl_string);

  for (byte b=0; b<255; b++){
    delay(500);
    etl_string.push_back((char)b);
    print_etl_string(etl_string);

    #ifdef PRINT_CHAR_BUFF
      print_cstring_as_hex(char_buffer);
    #endif
  }

  while(true){
  }
}
```

jerabaul29, 2021-02-15:
PS: I think I will just stick to defining the string in the ```etl::string<string_max_size> etl_string;``` way and use the ```data()``` member to get raw access to the data, but still would be interested to understand where the problem is :) .

jwellbelove, 2021-02-15:
Sorry, there's an error in the documentation example.

What you need is
```
char char_buffer[string_max_size + 1];
etl::string_ext etl_string(char_buffer, etl::size(char_buffer));
```

BTW You can simplify your print function by using a `const etl::istring&` as a parameter.

```
void print_etl_string(const etl::istring& etl_string){
```
and then you won't get multiple copies of the print function being generated.

jerabaul29, 2021-02-15:
Ok, many thanks for the help :) .

Yes, that is a very good point, thank you for pointing the change to a string ref :) .

I keep open just as a reminder, feel free to close once the documentation is updated.

closed 2021-02-15 by jwellbelove