ETLCPP/etl #335: how to "update / fix" a string after having modified it in low-level way through direct raw memory access

jerabaul29, 2021-02-15:
I have to provide raw memory pointers to some API functions that act on strings. These are "C-style" APIs, and I just have to adapt to them.

Providing the pointer to raw data array seems to be quite ok with the ```etl::string.data()``` member function. This allows the low-level functions to write to the raw data array.

However, I do not manage to get the string in a "consistent" way afterhand. Any idea how to make this work?

```cpp
#include "etl.h"
#include "etl/string.h"

void print_etl_string(const etl::istring& etl_string){
  Serial.println(F("--- etl string printout ---"));
  Serial.print(F("print etl string, max size "));
  Serial.print(etl_string.max_size());
  Serial.print(F(" crrt size: "));
  Serial.println(etl_string.size());

  Serial.print(F("content currently in use    : "));
  for (int i=0; i<etl_string.size(); i++){
    Serial.print("0x");
    Serial.print(etl_string.data()[i], HEX);
    Serial.print(F(" | "));
  }
  Serial.println();

  Serial.print(F("content currently not in use: "));
  for (int i=etl_string.size(); i<etl_string.max_size(); i++){
    Serial.print("0x");
    Serial.print(etl_string.data()[i], HEX);
    Serial.print(F(" | "));
  }
  Serial.println();
}

constexpr size_t string_max_size = 8;

etl::string<string_max_size> etl_string;

void print_cstring_as_hex(char * start){
  char * crrt_char = start;
  Serial.print(F("cstring as hex: "));

  while (true){
    Serial.print(F("0x"));
    Serial.print(*crrt_char, HEX);
    Serial.print(" | ");
    if (* crrt_char == '\0'){
      break;
    }
    crrt_char ++;
  }
  Serial.println();
}

void setup() {
  Serial.begin(115200);
  Serial.println(F("--- setup ---"));
  Serial.println();

  print_cstring_as_hex("hello");
  Serial.println();

  print_etl_string(etl_string);
  Serial.println();
  
  etl_string.data()[0] = 'h';
  etl_string.data()[1] = 'e';
  etl_string.data()[2] = 'l';
  etl_string.data()[3] = 'l';
  etl_string.data()[4] = 'o';
  print_cstring_as_hex(etl_string.data());

  // I look for a command to "update" the string so that it is "hello".
  // I.e, I would like to call a command, so that the string becomes:
  // content currently in use    : 0x68 | 0x65 | 0x6C | 0x6C | 0x6F |
  // content currently not in use: 0x0 | 0x0 | 0x0 
  // etl_string.resize(5);  // does not help: replace all my chars with some \0
  // etl_string.repair();   // does not help: nothing happens
  print_etl_string(etl_string);

  print_cstring_as_hex(etl_string.data());
}

void loop() {
}
```

This may also be relevant for other containers such as array. If you see another way around I will be very interested. Of course this is a toy example, but I have some functions that I would like to make work on your containers, while they only accept pointers to raw data.

jwellbelove, 2021-02-15:
What you need to do is to resize the string to `max_size()` before you pass it to the C API. Afterwards, you resize it back down to the real length of the string (+1), either from what the C API returns to you, or from from searching the string for the terminating null.

`repair()` is there to ensure that the internal pointers (for a non-external buffered string) are correct after a low level `memcpy()`.

jwellbelove, 2021-02-15:
Maybe fix the length with `etl_string.resize(etl::strlen(etl::string.data()));`?

jwellbelove, 2021-02-15:
I'm currently adding an `update_size()` member function to simplify this use case.

jerabaul29, 2021-02-15:
Excellent :) . Can we get this on other containers too which size can vary? (vectors, circular buffer, ...)? :)

jwellbelove, 2021-02-16:
It easy for strings, as they have a defined terminator value. Not so easy for other containers though.

jerabaul29, 2021-02-16:
Ok, I see. Would it be possible to give allow doing this "by hand'? Either giving the start and end of the raw memory data to set, or the new length to update to, or how many entries were added, or something like this? Otherwise it is quite hard to use for example the vector in combination to low level API :( .

jwellbelove, 2021-02-16:
That would only work on containers that have their data in a contiguous block, which would restrict it to vector.
deque and circular_buffer wrap around, so the C function would have to implement the rollover functionality correctly.
For circular_buffer there is the added issue of when old data is overwritten. Getting that the re-sync properly may be tricky.

With a vector you can use the resize(max_size()) => resize() method.

jwellbelove, 2021-02-16:
```
etl::vector<char, 100> buffer;

buffer.resize(buffer.max_size());

size_t charsAdded = AddData(buffer.data());

buffer.resize(charsAdded );
```

jwellbelove, 2021-02-16:
If you are adding to a vector.

```
etl::vector<char, 100> buffer;

buffer.resize(buffer.max_size());

size_t charsAdded = AddData(buffer.data() + buffer.size());

buffer.resize(buffer.size() + charsAdded);
```

jwellbelove, 2021-02-16:
Thinking about it, circular_buffer may not be _too_ difficult. I'll give it some thought.

jwellbelove, 2021-02-18:
Pushed to 20.3.0

closed 2021-02-18 by jwellbelove