ETLCPP/etl #336: Erroneous calculation of bit position in bitset method find_next

Chiraffollo, 2021-02-16:
The method find_next fails to return the correct position of a set bit in a bitset if the position to start from points to a byte in the data array that is empty (no match).

Example:
```
etl::bitset<24> bits;
bits.set(12);
std::cout << bits.find_first(true) << std::endl;
std::cout << bits.find_next(true, 4) << std::endl;

etl::bitset<16> bits2;
bits2.set(12);
std::cout << bits2.find_first(true) << std::endl;
std::cout << bits2.find_next(true, 4) << std::endl;
```
Output of sample program:
```
12
16
12
18446744073709551615
```
The problem is, that in bitset.h:611 the algorithm always increases `position` by `BITS_PER_ELEMENT` if there is no match in the byte. But `position` was initialized by variable `bit`. Therefore, `position` after first empty byte is `bit + BITS_PER_ELEMENT`.

Possible fix:
Replace bitset.h:611 by `position += (BITS_PER_ELEMENT - bit`

jwellbelove, 2021-02-17:
Thanks for the bug report. I'll look at that today.

jwellbelove, 2021-02-17:
Fixed in 20.2.3

closed 2021-02-17 by jwellbelove