ETLCPP/etl #337: Feature request: Forward list with push_back and pop_front modfiers

AbdManian, 2021-02-17:
In packet communication we are using a container that is Queue+linkedlist. Delete items from the middle of the list without expense of moving elements.
I think is a good idea to add such feature to the forward_list or define a new container.

Thanks,

jwellbelove, 2021-02-17:
The ETL's `forward_list` was designed to be a clone of the STL's `forward_list`.
`forward_list` does have a `pop_front`, but not a `push_back` or `pop_back`, as this would be very inefficient for this type of container.

`etl::list` would be more suitable for your needs.

AbdManian, 2021-02-17:
In case of no-shared pool we can perform `push_back` in O(1) (base class has a pointer to the first element). Is this correct?

I've considered using `et::list` but it uses much more memory as I actually using a linked-list in one direction.

jwellbelove, 2021-02-17:
> In case of no-shared pool we can perform push_back in O(1)

Which container are you talking about?

jwellbelove, 2021-02-17:
The issue with a forward list is that to do anything (without traversing the list from the beginning every time) you need to have a reference to the item _before_ the item you want to erase or push.

jwellbelove, 2021-02-17:
What type of item are you storing in the list?

AbdManian, 2021-02-17:
> 
> 
> What type of item are you storing in the list?
We are using list for storing some communication packets. We need queue like behavior but also we need deleting some items from middle of the list (some packets are rejected to transfer).
 

> 
> 
> > In case of no-shared pool we can perform push_back in O(1)
> 
> Which container are you talking about?

I mean `forward_list`

jwellbelove, 2021-02-17:
I'm sorry, I'm a bit confused. As `forward_list` doesn't have a `push_back`, how can you push in O(1)?
Or are you saying you _want_ to `push_back` in O(1)?

`forward_list` doesn't keep a pointer to the item last in the list, so `push_back`, if implemented, would be O(N).

To `pop_back`, it would need a pointer to one-before the last item. push_back could of course use this pointer too, via indirection through the 'next' pointer.

Erasing would still be an issue, as you need a reference to the item _before_ the one you want to erase.

You may find that `indirect_vector` may be a suitable compromise, as erasing an item only requires the copying of pointers and a release to a pool, which can be a fairly fast process.

jwellbelove, 2021-02-17:
Are you unable to check the packet _before_ placing it in the queue?

AbdManian, 2021-02-17:
> 
> 
> Are you unable to check the packet _before_ placing it in the queue?

Unfortunately is not possible. State of packets are provided after creation of the list.

jwellbelove, 2021-02-17:
Can you just test/discard the packet when it is reaches the front of the queue, or would that require an excessively long queue for your application?

AbdManian, 2021-02-17:
My current implementation is an linked list (chained from first to last element) with two pointers. One points to the start of the list and second to the last element. All free elements also chained in the link list. By deleting or popping from the back of the list the free element is added to the chain of free elements.
Pop from start of list and push to the end of list have O(1). Deleting from the middle of the list also has O(1) (I mean expense of joining not traveling from start of the list)

AbdManian, 2021-02-17:
> 
> 
> Can you just test/discard the packet when it is reaches the front of the queue, or would that require an excessively long queue for your application?

This application is somehow similar to ARQ algorithm. 
 * All packets in queue will be processes/transferred like vector of packets. 
 * New packets are added to the back of the queue
 * Discarded packets can be deleted from any position of the list
 * Order of packets in the list must be maintained

jwellbelove, 2021-02-17:
Unfortunately the ETL doesn't currently have a container with that sort functionality. That sounds like it could be a useful addition.
I assume when you are erasing, you already have a reference to the item before it, to get O(1)?

AbdManian, 2021-02-17:
> 
> 
> Unfortunately the ETL doesn't currently have a container with that sort functionality. That sounds like it could be a useful addition.
> I assume when you are erasing, you already have a reference to the item before it, to get O(1)?

Yes erase is preformed in traverse (similar to `remove_if`) and reference before it is available. 
If you guide me where to start I can contribute to create new (or extend existing) container (but I'm not very good at C++).

jwellbelove, 2021-02-17:
There is an `etl::intrusive_queue` that could possibly be extended with an erase function and wrapped in a linked_queue class with an `etl::pool`.

jwellbelove, 2021-02-17:
I'm going to close this, as I think we've covered the possible options

closed 2021-02-17 by jwellbelove