ETLCPP/etl #348: Using etl/string.h gives basic_string.h:95:26: error: ‘ETL_STRING_FILE_ID’ was not declared in this scope

bolry, 2021-03-11:
Just include <etl/string.h> and you get basic_string.h:95:26: error: ‘ETL_STRING_FILE_ID’ was not declared in this scope in C++98/03 mode.

Using etl_profile.h with content: `#include <etl/profiles/cpp03.h>`

g++-10  -I. -isystem etl-src/include -std=gnu++98 -c string.h.t.cpp

Similar problem with array_wrapper.h ==> error: ‘ETL_ARRAY_WRAPPER_FILE_ID’ was not declared in this scope

jwellbelove, 2021-03-11:
That's annoying, none of my compilers picked that up and neither did the Github CI!

bolry, 2021-03-11:
I've made one test with multiple files where each header has it's own cpp-file just to see if it still works in C++98/03 mode. It's been "rewarding". :-)

jovere, 2021-03-13:
It looks like this is all due to commit 87554af4f8a2b8743d729b911d5723b92f7a031e.  What I don't understand is that if I build the tests, it works fine, but if I just build a simple file with the following contents, it breaks just by including the header:
```c++
#include "etl/string.h"

int main()
{

}
```
Note: I first saw this when trying to use etl::strings in compiler explorer.
<details><summary>Compiler Output (expand)</summary>

```
$ gcc test.cpp -Ietl/include
In file included from etl/include/etl/array.h:45:0,
                 from etl/include/etl/basic_string.h:46,
                 from etl/include/etl/string.h:35,
                 from test.cpp:1:
etl/include/etl/basic_string.h: In constructor ‘etl::string_empty::string_empty(etl::exception::string_type, etl::exception::numeric_type)’:
etl/include/etl/basic_string.h:95:57: error: ‘ETL_STRING_FILE_ID’ was not declared in this scope
       : string_exception(ETL_ERROR_TEXT("string:empty", ETL_STRING_FILE_ID"A"), file_name_, line_number_)
                                                         ^
etl/include/etl/error_handler.h:305:53: note: in definition of macro ‘ETL_ERROR_TEXT’
   #define ETL_ERROR_TEXT(verbose_text, terse_text) (terse_text)   // Use the terse text.
                                                     ^~~~~~~~~~
etl/include/etl/basic_string.h:95:57: note: suggested alternative: ‘ETL_VARIANT_FILE_ID’
       : string_exception(ETL_ERROR_TEXT("string:empty", ETL_STRING_FILE_ID"A"), file_name_, line_number_)
                                                         ^
etl/include/etl/error_handler.h:305:53: note: in definition of macro ‘ETL_ERROR_TEXT’
   #define ETL_ERROR_TEXT(verbose_text, terse_text) (terse_text)   // Use the terse text.
                                                     ^~~~~~~~~~
etl/include/etl/basic_string.h:95:75: error: expected ‘)’ before string constant
       : string_exception(ETL_ERROR_TEXT("string:empty", ETL_STRING_FILE_ID"A"), file_name_, line_number_)
                                                                           ^
etl/include/etl/error_handler.h:305:53: note: in definition of macro ‘ETL_ERROR_TEXT’
   #define ETL_ERROR_TEXT(verbose_text, terse_text) (terse_text)   // Use the terse text.
                                                     ^~~~~~~~~~
In file included from etl/include/etl/string.h:35:0,
                 from test.cpp:1:
etl/include/etl/basic_string.h:97:5: error: no matching function for call to ‘etl::string_exception::string_exception()’
     }
     ^
etl/include/etl/basic_string.h:80:5: note: candidate: etl::string_exception::string_exception(etl::exception::string_type, etl::exception::string_type, etl::exception::numeric_type)
     string_exception(string_type reason_, string_type file_name_, numeric_type line_number_)
     ^~~~~~~~~~~~~~~~
etl/include/etl/basic_string.h:80:5: note:   candidate expects 3 arguments, 0 provided
etl/include/etl/basic_string.h:76:9: note: candidate: constexpr etl::string_exception::string_exception(const etl::string_exception&)
   class string_exception : public etl::exception
         ^~~~~~~~~~~~~~~~
etl/include/etl/basic_string.h:76:9: note:   candidate expects 1 argument, 0 provided
etl/include/etl/basic_string.h:76:9: note: candidate: constexpr etl::string_exception::string_exception(etl::string_exception&&)
etl/include/etl/basic_string.h:76:9: note:   candidate expects 1 argument, 0 provided
etl/include/etl/basic_string.h:97:5: error: expected ‘{’ at end of input
     }
     ^
In file included from etl/include/etl/array.h:45:0,
                 from etl/include/etl/basic_string.h:46,
                 from etl/include/etl/string.h:35,
                 from test.cpp:1:
etl/include/etl/basic_string.h: In constructor ‘etl::string_out_of_bounds::string_out_of_bounds(etl::exception::string_type, etl::exception::numeric_type)’:
etl/include/etl/basic_string.h:109:58: error: ‘ETL_STRING_FILE_ID’ was not declared in this scope
       : string_exception(ETL_ERROR_TEXT("string:bounds", ETL_STRING_FILE_ID"B"), file_name_, line_number_)
                                                          ^
etl/include/etl/error_handler.h:305:53: note: in definition of macro ‘ETL_ERROR_TEXT’
   #define ETL_ERROR_TEXT(verbose_text, terse_text) (terse_text)   // Use the terse text.
                                                     ^~~~~~~~~~
etl/include/etl/basic_string.h:109:58: note: suggested alternative: ‘ETL_VARIANT_FILE_ID’
       : string_exception(ETL_ERROR_TEXT("string:bounds", ETL_STRING_FILE_ID"B"), file_name_, line_number_)
                                                          ^
etl/include/etl/error_handler.h:305:53: note: in definition of macro ‘ETL_ERROR_TEXT’
[... 86 lines cut ...]
                                                                              ^
etl/include/etl/error_handler.h:305:53: note: in definition of macro ‘ETL_ERROR_TEXT’
   #define ETL_ERROR_TEXT(verbose_text, terse_text) (terse_text)   // Use the terse text.
                                                     ^~~~~~~~~~
In file included from etl/include/etl/string.h:35:0,
                 from test.cpp:1:
etl/include/etl/basic_string.h:139:5: error: no matching function for call to ‘etl::string_exception::string_exception()’
     }
     ^
etl/include/etl/basic_string.h:80:5: note: candidate: etl::string_exception::string_exception(etl::exception::string_type, etl::exception::string_type, etl::exception::numeric_type)
[... cut ...]
etl/include/etl/basic_string.h:76:9: note: candidate: constexpr etl::string_exception::string_exception(const etl::string_exception&)
   class string_exception : public etl::exception
         ^~~~~~~~~~~~~~~~
etl/include/etl/basic_string.h:76:9: note:   candidate expects 1 argument, 0 provided
etl/include/etl/basic_string.h:76:9: note: candidate: constexpr etl::string_exception::string_exception(etl::string_exception&&)
etl/include/etl/basic_string.h:76:9: note:   candidate expects 1 argument, 0 provided
etl/include/etl/basic_string.h:139:5: error: expected ‘{’ at end of input
     }
     ^
```

</details>

jwellbelove, 2021-03-13:
I'm working on a fix this weekend + adding bolry's sanity check tests.

jwellbelove, 2021-03-14:
Fixed in 20.6.2

closed 2021-03-14 by jwellbelove