ETLCPP/etl #355: Bug: std::sort on etl::deque breaks with MSVC v19.28

dingari, 2021-03-23:
I've just updated MSVC to v19.28 and I'm getting an error when attempting to sort an `etl::deque` with `std::sort`. This is code that was compiling on MSVC v19.27.

```
#include <algorithm>
#include <etl/deque.h>

int main()
{
    etl::deque<int, 64> d{};

    std::sort(d.begin(), d.end());
}
```

Even this simple example will fail to compile with an error from the compiler saying that it can't assign to a const variable (error C3892). 

In my case, it's a deque of structs, so I get a slightly different error relating to the structs constructors (C2678), but the net result is the same - not being able to assign to a const variable.

```
error C2678: binary '=': no operator found which takes a left-hand operand of type 'const T' (or there is no acceptable conversion)
        with
        [
            T=MyStruct
        ]
<source path omitted>: note: could be 'MyStruct &MyStruct::operator =(MyStruct &&)'
<source path omitted>: note: or       'MyStruct &MyStruct::operator =(const MyStruct &)'
```

This might be due to some changes in the MSVC STL implementation, but if I change to a std::deque, it compiles fine.

ETLCPP is not yet supported on Compiler Explorer for MSVC compilers, but I submitted [a request](https://github.com/compiler-explorer/compiler-explorer/issues/2514) earlier today. Having ETLCPP supported for the major compilers there will be a huge help.

jwellbelove, 2021-03-25:
It looks like the operators in most of the ETL's container's `iterator` were mistakenly returning `const_iterator` instead of `iterator`.
I'm in the process of making the changes.

jwellbelove, 2021-03-26:
Fix pushed to 20.7.0

closed 2021-03-26 by jwellbelove