ETLCPP/etl #358: Issue compiling (clang & cmake)

part22, 2021-03-25:
Hi,

I am using WSL and clang to compile a binary for Arduino. I am including `<etl/bitset.h>` but when I try to compile I am getting the following error:
```
/home/part22/avr-llvm/llvm-project/build/bin/clang++ --target=avr -DARDUINO=10813 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DF_CPU=16000000L -I/home/part22/projects/arduino-clang/source/include -isystem /home/part22/projects/arduino-clang/source/third-party/etl/include -mmcu=atmega328p -nostdlib -Os -DNDEBUG -fno-lto -fdata-sections -fno-exceptions -ffunction-sections -fno-threadsafe-statics -std=c++17 -MD -MT CMakeFiles/blinky.dir/blink.cpp.obj -MF CMakeFiles/blinky.dir/blink.cpp.obj.d -o CMakeFiles/blinky.dir/blink.cpp.obj -c /home/part22/projects/arduino-clang/source/blink.cpp
In file included from /home/part22/projects/arduino-clang/source/blink.cpp:1:
In file included from /home/part22/projects/arduino-clang/source/third-party/etl/include/etl/bitset.h:43:
/home/part22/projects/arduino-clang/source/third-party/etl/include/etl/integral_limits.h:123:39: error: in-class initializer for static data member is not a constant expression
    static const unsigned short max = USHRT_MAX;
                                      ^~~~~~~~~
/home/part22/avr-llvm/llvm-project/build/lib/clang/12.0.0/include/limits.h:55:33: note: expanded from macro 'USHRT_MAX'
#define USHRT_MAX (__SHRT_MAX__ *2  +1)
                  ~~~~~~~~~~~~~~^~~~~~~
```

I am not sure what I did wrong here. If anyone has had the same issue, I would be glad to know how I can solve this.
I am happy to provide more details if needed.

Cheers!

jwellbelove, 2021-03-25:
I would have though that `(__SHRT_MAX__ *2  +1)` would be a constant expression.
Do you know what the definition of `__SHRT_MAX__` is?

jwellbelove, 2021-03-25:
Can you try this modified ETL `integral_limits.h` file to see if it fixes the issue?

[integral_limits.zip](https://github.com/ETLCPP/etl/files/6203524/integral_limits.zip)

part22, 2021-03-26:
I also thought that `(__SHRT_MAX__ *2 +1)` was a constant expression but apparently I was wrong.
I was not able to find the definition of `__SHRT_MAX__`.

When I use the modified `integral_limits` header I get that error:
```
/home/part22/avr-llvm/llvm-project/build/bin/clang++ --target=avr -DARDUINO=10813 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DF_CPU=16000000L -I/home/part22/projects/arduino-clang/source/include -isystem /home/part22/projects/arduino-clang/source/third-party/etl/include -mmcu=atmega328p -nostdlib -Os -DNDEBUG -fno-lto -fdata-sections -fno-exceptions -ffunction-sections -fno-threadsafe-statics -std=c++17 -MD -MT CMakeFiles/blinky.dir/blink.cpp.obj -MF CMakeFiles/blinky.dir/blink.cpp.obj.d -o CMakeFiles/blinky.dir/blink.cpp.obj -c /home/part22/projects/arduino-clang/source/blink.cpp
In file included from /home/part22/projects/arduino-clang/source/blink.cpp:1:
In file included from /home/part22/projects/arduino-clang/source/third-party/etl/include/etl/bitset.h:43:
/home/part22/projects/arduino-clang/source/third-party/etl/include/etl/integral_limits.h:123:40: error: constexpr variable 'max' must be initialized by a constant expression
    static ETL_CONSTANT unsigned short max = USHRT_MAX;
                                       ^     ~~~~~~~~~
/home/part22/projects/arduino-clang/source/third-party/etl/include/etl/integral_limits.h:123:46: note: value 65534 is outside the range of representable values of type 'int'
    static ETL_CONSTANT unsigned short max = USHRT_MAX;
                                             ^
/home/part22/avr-llvm/llvm-project/build/lib/clang/12.0.0/include/limits.h:55:33: note: expanded from macro 'USHRT_MAX'
#define USHRT_MAX (__SHRT_MAX__ *2  +1)
                                ^
```
One thing I can think of is that I have compiled clang from source to enable avr support. Maybe the `limits.h` from clang is not good for avr microcontroller?

jwellbelove, 2021-03-26:
Looks like it still doesn't think that __SHRT_MAX__ is a constant expression.

GatorQue, 2021-03-27:
I did a quick Google search for `__SHRT_MAX__` and found this link from GCC which explicitly says not to use this macro as it may not be available:
https://gcc.gnu.org/onlinedocs/cpp/Common-Predefined-Macros.html

Perhaps its a GCC specific macro definition? Should we try to find a different solution to compute this const expression that works for GCC and CLANG?

GatorQue, 2021-03-27:
Looking at the error posted above more closely it seems that the complaint is in the clang version of `limits.h`. Are we somehow including the file out of order? Perhaps it was meant to be included from a parent header instead?

part22, 2021-03-27:
Hi @GatorQue,

What I found strange is that in the `integral_limits` header, the max limit for `unsigned char` is also defined in terms of another macro but it is not causing any issues.

That is the definition of the limits for `unsigned char` in `integral_limits`:
```cpp
  template <>
  struct integral_limits<unsigned char>
  {
    static const unsigned char min = 0;
    static const unsigned char max = UCHAR_MAX;
    static const int bits          = CHAR_BIT;
    static const bool is_signed    = etl::is_signed<unsigned char>::value;
  }
```
`UCHAR_MAX` expands to `(__SCHAR_MAX__*2  +1)` as defined in clang's `limits` header.
```cpp
#define UCHAR_MAX (__SCHAR_MAX__*2  +1)
```

part22, 2021-03-27:
Also when I mouse over `__SHRT_MAX__` it shows the value 32767 so, it should be available. As a test, I hardcoded the value and it solves the problem. However, I now have other issues, not related to `integral_limits`.

GatorQue, 2021-03-27:
I have WSL installed on my computer, what OS do you have running under WSL and can you give me steps to setup a similar environment?

jwellbelove, 2021-03-27:
I've been running WSL with Ubuntu 20.03 on my machine.

GatorQue, 2021-03-27:
That is what I have installed too, so that is good. From my research it doesn't sound like AVR support in clang is particularly mature at this moment, I wonder if that has any bearing on things. I'm hoping to try to create the smallest example possible to demonstrate the error and see if we can narrow it down to CLANG or ETL or the program being written.

part22, 2021-03-27:
I am using WSL 2 and Ubuntu 20.04, I compiled clang from source to enable avr support and then tried to compile my program with it. I am going to try to make a small repro case. Is it ok to make a git repo for the repro case and link it here?

GatorQue, 2021-03-27:
I think that might be helpful.

part22, 2021-04-04:
Here is the link to the minimal repro case: [https://github.com/part22/etl-repro-issue-358](url)
If you need any help, feel free to let me know.

GatorQue, 2021-04-05:
I'm working on the clang requirement, do I need to build clang, llvm, or both from sources?

part22, 2021-04-06:
If I remember correctly I did:
```
git clone https://github.com/llvm/llvm-project.git --depth=1
cd llvm-project
cmake -Sllvm -Bbuild -G"Ninja" -DLLVM_ENABLE_PROJECTS="clang" -DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD=AVR -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=[installation-path default is usr/local]
ninja -Cbuild
```

GatorQue, 2021-04-09:
I followed the steps you outlined along with

part22, 2021-04-09:
I believe it's in binutils-avr, so this should sort it out:
`sudo apt-get install binutils-avr `

I appreciate your help on this, cheers :)

GatorQue, 2021-04-11:
OK, I think I have all the dependencies now, but it won't compile correctly. Here is the output I am seeing:
```
$ ninja -C build
ninja: Entering directory `build'
[1/2] Building CXX object CMakeFiles/etl-repro.dir/main.cpp.obj
FAILED: CMakeFiles/etl-repro.dir/main.cpp.obj
/usr/local/bin/clang++ --target=avr -DARDUINO=10813 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DF_CPU=16000000L -I/home/rlindeman/etl-issue/etl-repro-issue-358/source/include -isystem /home/rlindeman/etl-issue/etl-repro-issue-358/source/third-party/etl/include -mmcu=atmega328p -nostdlib -Os -DNDEBUG -fno-lto -fdata-sections -fno-exceptions -ffunction-sections -fno-threadsafe-statics -std=c++17 -MD -MT CMakeFiles/etl-repro.dir/main.cpp.obj -MF CMakeFiles/etl-repro.dir/main.cpp.obj.d -o CMakeFiles/etl-repro.dir/main.cpp.obj -c /home/rlindeman/etl-issue/etl-repro-issue-358/source/main.cpp
In file included from /home/rlindeman/etl-issue/etl-repro-issue-358/source/main.cpp:1:
In file included from /home/rlindeman/etl-issue/etl-repro-issue-358/source/third-party/etl/include/etl/bitset.h:34:
/usr/include/string.h:26:10: fatal error: 'bits/libc-header-start.h' file not found
#include <bits/libc-header-start.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.
```
Any thoughts?

part22, 2021-04-11:
I had the same issue, this fixed it:
```
sudo apt-get install gcc-multilib
```

GatorQue, 2021-04-12:
I am able to reproduce the issue now, I will try to find some time to investigate. Thanks!

jwellbelove, 2021-05-14:
Has there been any updates to this?

GatorQue, 2021-05-14:
I'm sorry but I haven't had a minute to look further into this.

part22, 2021-06-18:
I had a bit more time to look into the issue and I found that I did not configure my build correctly. It was using the wrong standard C library. Once I fixed that, I still had an issue but I think I found a fix for it.

I have made a pull request for it: #389.

closed 2021-06-18 by part22