ETLCPP/etl #360: warning: floating constant exceeds range for AVR

jputcu, 2021-04-02:
Wanting to use <etl/span.h> I got lots of errors about **Nan** and **huge_val** with my AVR build using the official Arduino avr-gcc compiler (7.3).
ETL version is 19.5.0
Zo I activated the workarrounds using: **-DETL_NO_CPP_NAN_SUPPORT -DETL_NO_HUGE_VAL_SUPPORT**. 
Now compilation is ok with lots of warnings about exceeded ranges:
~~~~
etl\limits.h(548,5): warning: floating constant exceeds range of 'float' [-Woverflow]
		     static float infinity() { return HUGE_VALF; }
etl\limits.h(574,5): warning: floating constant exceeds range of 'double' [-Woverflow]
		     static double infinity() { return HUGE_VAL; }
etl\limits.h(600,5): warning: floating constant exceeds range of 'long double' [-Woverflow]
		     static long double infinity() { return HUGE_VALL; }
~~~~

jwellbelove, 2021-04-02:
A quick fix would be to define your own HUGE_VALF, HUGE_VAL and HUGE_VALL as FLT_MAX, DBL_MAX and LDBL_MAX respectively.

jputcu, 2021-04-02:
You are right, but I can live with the compiler warnings until there is a solution.

jwellbelove, 2021-04-02:
That's what I'll probably do in limits.h anyway.

jwellbelove, 2021-04-14:
Fixed in 20.9.0

closed 2021-04-14 by jwellbelove

jputcu, 2021-04-14:
I'm now having issues with missing the DBL_MAX not being defined.
Also, it appears their sizes can depend on compiler configuration: https://gcc.gnu.org/onlinedocs/gcc-10.2.0/gcc/AVR-Options.html *-mdouble=bits* and *-mlong-double=bits*.
Getting a dump from my cross compiler defines I get the following interesting:
~~~~~
#define __DBL_DENORM_MIN__ double(1.40129846e-45L)
#define __SIZEOF_LONG_DOUBLE__ 4
#define __DBL_MAX__ double(3.40282347e+38L)
#define __DBL_MIN__ double(1.17549435e-38L)
#define __SIZEOF_DOUBLE__ 4
~~~~~