ETLCPP/etl #361: Use FSM as a hierarchical state machine

jovere, 2021-04-07:
Is there any way to use the FSM as a hierarchical state machine?  I have a fairly complex state machine that I would like to create, which has some overall exit/entry actions and transitions I would like to create in a "parent" state.  For example, if I have a system fault, I'd like most, but not all of the states to go back to a safe "idle" state.  Has anything been done like that with the FSM?

Thanks,
Jeremy

jwellbelove, 2021-04-07:
Someone did push an HSM implementation to a feature branch some time ago, but I want to explore various other implementations first be I commit the ETL to one.

jeremyerdmann, 2021-04-08:
A hierarchical option would be a great add to the library.  I've got a teammate working on a solution using the FSM.  Our state diagram included 4 substates under one of the states.  Plan is to break it out into 4 distinct states, but the more we discuss the design, the more states we're adding.  Our diagnostic override mode feels like a case where adding it could have been a pair of lines on the diagram vs. a pair to every state in the system.  It sure can add up quick when things get complex!

aboseley, 2021-04-16:
Why not use boost::msm?
https://www.boost.org/doc/libs/1_76_0/libs/msm/doc/HTML/index.html

jovere, 2021-04-22:
> Why not use boost::msm?
> https://www.boost.org/doc/libs/1_76_0/libs/msm/doc/HTML/index.html

Because I would like to minimize the number of libraries I need to use in an embedded system.
I have created a PR that implements an `hfsm` by inheriting from and slightly modifying the `fsm` in #374.  I have also added a test based off of the `fsm` test.

jwellbelove, 2021-04-22:
I look forward to seeing your solution.

closed 2021-05-14 by jwellbelove