ETLCPP/etl #365: etl::map with etl::string as key and value

fabssor, 2021-04-12:
Hello,

I am trying to create a map with a string as key and value. With STL this is possible see here: https://www.cplusplus.com/reference/map/map/at/
or here:
https://en.cppreference.com/w/cpp/container/map

I tried the following with ETL: 
```cpp
etl::map<etl::string<20>, etl::string<30>, 2> map;

etl::array<const char[20], 2> keys= { 
            "key",
            "otherKey", 
        };
for(uint8_t i = 0; i < headers.size(); ++i )
{
     httpHeaders[etl::make_string(keys[i])].assign(cString, strlen(cString)); /* assign the values*/
}
```
This seems to work but when I try to assign the map to another map things seem to break.

```cpp
/* Constructor of other class */
someOtherClass(etl::imap<etl::string<20>, etl::string<30>>& imap) :
    memberMap{imap.begin(), imap.end()}
{};

/* member variable defined as following: */
etl::map<etl::string<20>, etl::string<30>, 2> memberMap;
```

I can not access both keys in the map. Also tried different ways to init the `memberMap` but did not succeed. 
```cpp
/* this works */
for(auto elem : map)
{
        httpHeaders_[elem.first] = elem.second;
        std::cout << ">> " << elem.first.data() << ": " << elem.second.data() << std::endl;
}   
/* this only for one element*/
etl::array<const char[20], 2> keys= { 
            "key",
            "otherKey", 
        };
std::cout << "Authorization: " << httpHeaders.at(etl::make_string(keys[0])).data() << std:: endl;
std::cout << "Cookie: " << httpHeaders.at(etl::make_string(keys[1])).data() << std:: endl; /* Segmentation fault here */
/* This does not work at all */
etl::array<const char[20], 2> keys= { 
etl:string<19> key1("key");
etl:string<19> key2("otherKey");

std::cout << "Authorization: " << httpHeaders.at(key1).data() << std:: endl;
std::cout << "Cookie: " << httpHeaders.at(key2).data() << std:: endl; /* Segmentation fault here */

```
Using the []-operator does also not work.

Is this even possible to use a string as key? Thank you!

jwellbelove, 2021-04-12:
The creation of tag 20.40.1 was a mistake and was removed a while ago.
The current version is 20.8.0.

jwellbelove, 2021-04-12:
There is a bug in `etl::make_string` where the length of the string is set to the array size and not the length of the actual string _in_ the array.

jwellbelove, 2021-04-12:
For an immediate fix, replace the following in `etl/string.h`
Lines 474 to 490

```
  //***************************************************************************
  /// Make string from string literal or array
  //***************************************************************************
  template<size_t ARRAY_SIZE>
  etl::string<ARRAY_SIZE - 1> make_string(const char(&text)[ARRAY_SIZE])
  {
    return etl::string<ARRAY_SIZE - 1>(text, etl::strlen(text));
  }

  //***************************************************************************
  /// Make string with max capacity from string literal or array
  //***************************************************************************
  template<const size_t MAX_SIZE, const size_t SIZE>
  etl::string<MAX_SIZE> make_string_with_capacity(const char(&text)[SIZE])
  {
    return etl::string<MAX_SIZE>(text, etl::strlen(text));
  }
```

fabssor, 2021-04-13:
Thank you @jwellbelove looking forwared for the fix ! :)

jwellbelove, 2021-04-13:
Fixed v20.9.0

closed 2021-04-13 by jwellbelove