ETLCPP/etl #368: Trouble compiling code in platformio with arduino (esp8266)

gnalbandian, 2021-04-16:
Hi, I am tryingto test this library in my environement. 
I am using platformio with arduino for ESP8266.

I 've added to my platformio.ini lib_deps:
`etlcpp/Embedded Template Library ETL @ ^20.2.2`

And to the main.cpp
`#include <etl/circular_buffer.h>`

Which is the module I intend to use.

When trying to compile, without using any function or instantiation of circular_buffer I get all these errors:

> Compiling .pio\build\esp07\src\main.cpp.o
> In file included from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/algorithm.h:43:0,
>                  from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/vector.h:41,
>                  from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/circular_buffer.h:35,
>                  from src\main.cpp:101:
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:823:30: error: expected type-specifier
>    using remove_reference_t = std::remove_reference_t<T>;
>                               ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:833:28: error: expected type-specifier
>    using remove_pointer_t = std::remove_pointer_t<T>;
>                             ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:843:25: error: expected type-specifier
>    using add_pointer_t = std::add_pointer_t<T>;
>                          ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:863:26: error: expected type-specifier
>    using remove_const_t = std::remove_const_t<T>;
>                           ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:873:23: error: expected type-specifier
>    using add_const_t = std::add_const_t<T>;
>                        ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:893:29: error: expected type-specifier
>    using remove_volatile_t = std::remove_volatile_t<T>;
>                              ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:903:26: error: expected type-specifier
>    using add_volatile_t = std::add_volatile_t<T>;
>                           ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:913:23: error: expected type-specifier
>    using remove_cv_t = std::remove_cv_t<T>;
>                        ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:923:20: error: expected type-specifier
>    using add_cv_t = std::add_cv_t<T>;
>                     ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:1200:25: error: expected type-specifier
>    using make_signed_t = std::make_signed_t<T>;
>                          ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:1210:27: error: expected type-specifier
>    using make_unsigned_t = std::make_unsigned_t<T>;
>                            ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:1220:23: error: expected type-specifier
>    using enable_if_t = std::enable_if_t<B, T>;
>                        ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:1241:27: error: expected type-specifier
>    using remove_extent_t = std::remove_extent_t<T>;
>                            ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:1251:32: error: expected type-specifier
>    using remove_all_extents_t = std::remove_all_extents_t<T>;
>                                 ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:1271:19: error: expected type-specifier
>    using decay_t = std::decay_t<T>;
>                    ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:1299:34: error: expected type-specifier
>    using add_lvalue_reference_t = std::add_lvalue_reference_t<T>;
>                                   ^
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/type_traits.h:1310:34: error: expected type-specifier
>    using add_rvalue_reference_t = std::add_rvalue_reference_t<T>;
>                                   ^
> In file included from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/memory.h:41:0,
[... 46 lines cut ...]
>                  from src\main.cpp:45:
> c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\new:115:13: error: 'void operator delete(void*, void*)' previously defined here
>  inline void operator delete  (void*, void*) _GLIBCXX_USE_NOEXCEPT { }
>              ^
> In file included from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/memory.h:41:0,
>                  from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/vector.h:44,
>                  from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/circular_buffer.h:35,
>                  from src\main.cpp:101:
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h: In function 'void operator delete [](void*, void*)':
> .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h:45:13: error: redefinition of 'void operator delete [](void*, void*)'
>  inline void operator delete[](void*, void*) ETL_NOEXCEPT {}
>              ^
> In file included from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\ext\new_allocator.h:33:0,
>                  from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits\c++allocator.h:33,
>                  from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\allocator.h:46,
>                  from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\memory:63,
>                  from C:\Users\gonza\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:24,
>                  from src\main.cpp:45:
> c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\new:116:13: error: 'void operator delete [](void*, void*)' previously defined here
>  inline void operator delete[](void*, void*) _GLIBCXX_USE_NOEXCEPT { }
>              ^
> *** [.pio\build\esp07\src\main.cpp.o] Error 1

Any clues on how to get thisworking in platformio?
Thanks!

jwellbelove, 2021-04-16:
Some of those errors look like the ETL is configured for >=C++14, but the compiler does not support the `std::remove_reference_t<T>` style statements.

gnalbandian, 2021-04-16:
Thanks @jwellbelove, but I don't know what to do with your answer.
Is there a way around this errors? 
Sorry for the hassle..

jwellbelove, 2021-04-18:
Do you know what C++ standard you are compiling for? C++11? C++14?
Are you using etl_profile.h? If so, what do you have in it?

gnalbandian, 2021-04-19:
It's C++11

This is my etl_profile.h
```
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__
#define ETL_THROW_EXCEPTIONS
#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP
#define ETL_CPP11_SUPPORTED 
#endif
```

jwellbelove, 2021-04-19:
Try adding this

#define ETL_CPP11_SUPPORTED  1
#define ETL_CPP14_SUPPORTED  0
#define ETL_CPP17_SUPPORTED  0

gnalbandian, 2021-04-19:
Sorry, but still the same error

```
In file included from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/memory.h:41:0,
                 from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/vector.h:44,
                 from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/circular_buffer.h:35,
                 from src\main.cpp:104:
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h: In function 'void* operator new(size_t, void*)':
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h:41:14: error: redefinition of 'void* operator new(size_t, void*)'
 inline void* operator new(size_t, void* p) { return p; }
              ^
In file included from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\ext\new_allocator.h:33:0,
                 from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits\c++allocator.h:33,
                 from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\allocator.h:46,
                 from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\memory:63,
                 from C:\Users\gonza\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:24,
                 from src\main.cpp:46:
c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\new:109:14: error: 'void* operator new(std::size_t, void*)' previously defined here  
 inline void* operator new(std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
              ^
In file included from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/memory.h:41:0,
                 from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/vector.h:44,
                 from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/circular_buffer.h:35,
                 from src\main.cpp:104:
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h: In function 'void* operator new [](size_t, void*)':
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h:42:14: error: redefinition of 'void* operator new [](size_t, void*)'
 inline void* operator new[](size_t, void* p) { return p; }
              ^
In file included from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\ext\new_allocator.h:33:0,
                 from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits\c++allocator.h:33,
                 from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\allocator.h:46,
                 from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\memory:63,
                 from C:\Users\gonza\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:24,
                 from src\main.cpp:46:
c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\new:111:14: error: 'void* operator new [](std::size_t, void*)' previously defined here
 inline void* operator new[](std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
              ^
In file included from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/memory.h:41:0,
                 from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/vector.h:44,
                 from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/circular_buffer.h:35,
                 from src\main.cpp:104:
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h: In function 'void operator delete(void*, void*)':
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h:44:13: error: redefinition of 'void operator delete(void*, void*)'
 inline void operator delete(void*, void*) ETL_NOEXCEPT {}
             ^
In file included from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\ext\new_allocator.h:33:0,
                 from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits\c++allocator.h:33,
                 from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\allocator.h:46,
                 from c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\memory:63,
                 from C:\Users\gonza\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:24,
                 from src\main.cpp:46:
c:\users\gonza\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\new:115:13: error: 'void operator delete(void*, void*)' previously defined here
 inline void operator delete  (void*, void*) _GLIBCXX_USE_NOEXCEPT { }
             ^
In file included from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/memory.h:41:0,
                 from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/vector.h:44,
                 from .pio\libdeps\esp07\Embedded Template Library ETL\src/etl/circular_buffer.h:35,
                 from src\main.cpp:104:
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h: In function 'void operator delete [](void*, void*)':
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/placement_new.h:45:13: error: redefinition of 'void operator delete [](void*, void*)'
[... cut ...]
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/error_handler.h:278:51: error: exception handling disabled, use -fexceptions to enable
     #define ETL_ASSERT(b, e) {if (!(b)) {throw((e));}}                                     // If the condition fails, throws an exception.
                                                   ^
.pio\libdeps\esp07\Embedded Template Library ETL\src/etl/private/pvoidvector.h:193:7: note: in expansion of macro 'ETL_ASSERT'
       ETL_ASSERT(new_size <= CAPACITY, ETL_ERROR(vector_full));
       ^
*** [.pio\build\esp07\src\main.cpp.o] Error 1
```

jwellbelove, 2021-04-19:
It looks like it is seeing the standard library 'new' operator as well as the ETL's version.

Try adding 
#define ETL_USING_STD_NEW 1

gnalbandian, 2021-04-19:
Thanks @jwellbelove. That finally did the trick

Here is the profile I'm using for ESP8266 running arduino in platformio.

```
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__
#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP
#define ETL_USING_STD_NEW 1
#define ETL_COMPILER_GCC
#define ETL_CPP11_SUPPORTED 1
#define ETL_CPP14_SUPPORTED 1
#define ETL_CPP17_SUPPORTED 0
```

closed 2021-09-04 by jwellbelove