ETLCPP/etl #371: pool.h for uint8_t and other simple types

Revoltowiec, 2021-04-19:
Hello!

Background: 
I'm currently working on same simple generic wrapper for your etl::pool class. 
Main goal is to safely handling size and nullptr related constrains without throwing any exception from the library.
eg;

```
  etl::pool<MEMORY_ELEMENT, NUMBER_OF_ELEMENTS> memPool{};

  etl::optional<MEMORY_ELEMENT*> allocate()
  {
    if(memPool.full())
    {
      return {};
    }

    MEMORY_ELEMENT* memory = memPool.allocate();
    ENSURE(memory != nullptr);

    return { memory };
  }

  bool deallocate(const MEMORY_ELEMENT* const elementPtr)
  {
    REQUIRE(elementPtr != nullptr);

    if(memPool.is_in_pool(elementPtr))
    {
      memPool.release(elementPtr);
      return true;
    }
    return false;
  }
```

Problem:
Solution from above works fine when MEMORY_ELEMENT is a class type, but for releasing any integral/floating/array type I get an error from static_assert from pool.h :159.

problem method:
```
    //*************************************************************************
    /// Releases the object.
    /// Undefined behaviour if the pool does not contain a 'U' object derived from 'U'.
    /// \param p_object A pointer to the object to be destroyed.
    //*************************************************************************
    template <typename U>
    void release(const U* const p_object)
    {
      ETL_STATIC_ASSERT((etl::is_base_of<U, T>::value), "Pool does not contain this type");
      base_t::release(p_object);
    }

```

Finally question 😄  :

Limitation of this memory pool for class usage is intentional?

One of possible workarounds could be implementing template specialization of my wrapper for integral/floating/array types using vector, but maybe do you have some others ideas?

thanks!

jwellbelove, 2021-04-19:
No, that wasn't meant to be intentional. I should be able to fix that with a simple change to the ETL_ASSERT line.

Revoltowiec, 2021-04-20:
It would be great. 
May I ask, when this change could be available in master branch?

jwellbelove, 2021-04-20:
This morning.

jwellbelove, 2021-04-20:
Fixed in 20.9.2

closed 2021-04-20 by jwellbelove