ETLCPP/etl #372: Using wrong DBL_MAX compiler defines for AVR gcc

jputcu, 2021-04-20:
With latest version, I'm having issues with missing the DBL_MAX not being defined.
It appears their sizes can depend on compiler configuration: https://gcc.gnu.org/onlinedocs/gcc-10.2.0/gcc/AVR-Options.html *-mdouble=bits* and *-mlong-double=bits*.
Getting a dump from my cross compiler defines I get the following interesting:
~~~~~
#define __DBL_DENORM_MIN__ double(1.40129846e-45L)
#define __SIZEOF_LONG_DOUBLE__ 4
#define __DBL_MAX__ double(3.40282347e+38L)
#define __DBL_MIN__ double(1.17549435e-38L)
#define __SIZEOF_DOUBLE__ 4
~~~~~
There are more related variables, so I've attached the output of both Arduino (g++ 7.3) and Atmel Studio (g++5.4) g++:
[avr-gcc-vars_atmel.txt](https://github.com/ETLCPP/etl/files/6310739/avr-gcc-vars_atmel.txt)
[avr-gcc-vars_arduino.txt](https://github.com/ETLCPP/etl/files/6310740/avr-gcc-vars_arduino.txt)

jwellbelove, 2021-04-20:
Does AVR GCC not supply `float.h`?
Or does it not actually support `double`?

jwellbelove, 2021-04-20:
If it's just the macros that are missing then maybe I can add something like this.

```
#if defined(ETL_COMPILER_GCC) && defined(__AVR)
  #if !defined(DBL_MAX)
    #define DBL_MAX __DBL_MAX__
  #endif
#endif
```

jwellbelove, 2021-04-20:
It would be interesting to see what the GCC AVR `float.h` contains.

jwellbelove, 2021-04-20:
If I download a version of GCC AVR (https://blog.zakkemble.net/avr-gcc-builds/comment-page-1/) its `float.h` already defines `DBL_MAX` in terms of `__DBL_MAX__`

```
/* Maximum representable finite floating-point number,

	(1 - b**-p) * b**emax
*/
#undef FLT_MAX
#undef DBL_MAX
#undef LDBL_MAX
#define FLT_MAX		__FLT_MAX__
#define DBL_MAX		__DBL_MAX__
#define LDBL_MAX	__LDBL_MAX__
```

jputcu, 2021-04-20:
I've checked all toolchains, and indeed they have the "float.h" with the defines. So it is perhaps a matter of including this file a the right place.

jputcu, 2021-04-21:
Perhaps it's possible to avoid the inclusion of "integral_limits.h" from "span.h"?

jwellbelove, 2021-04-21:
Looking at `limits.h` I think that the checks for the floating point macros should **not** be inside the `#if ETL_NOT_USING_STL`.
Can you try the attached `limits.h` file?

[limits.zip](https://github.com/ETLCPP/etl/files/6350385/limits.zip)

jputcu, 2021-04-22:
Tried but still get the same error. Strange thing is that `float.h`, which is included by the `limits.h` contains the define!

jwellbelove, 2021-04-22:
Have you tried including `float.h` and used DBL_MAX in another file?

jputcu, 2021-04-23:
Found the issue, `DBL_MAX` is spelled incorrectly: `DLB_MAX`

jwellbelove, 2021-04-23:
Hah! 
Hidden in plain site.

closed 2021-04-30 by jputcu