ETLCPP/etl #376: ETL containers support for ranges in C++20?

shreyasbharath, 2021-05-13:
Hi there,

The below example compiles
```
    std::map<int, int> container;
    for (const auto kvp : container | std::views::values)
    {
    }
```

However, using an ETL container does not
```
    etl::map<int, int, 20> container;
    for (const auto kvp : container | std::views::values)
    {
    }
```

Is there any plan to support ranges in C++20?

Thanks.

jwellbelove, 2021-05-14:
It's on my todo list now.
I'll have to investigate how std::views are implemented, so that I can try to back-port them to the ETL for use with the No STL configuration.

jwellbelove, 2021-05-15:
As I've not played with ranges and views yet, I thought I'd try a simple example, based on your code.

This compiles and run fine for me!

```
#include <ranges>
#include <map>
#include <iostream>

#include "etl/map.h"

int main()
{
  std::map<int, int> container = { { 0, 0 },
                                   { 1, 1 },
                                   { 2, 2 },
                                   { 3, 3 },
                                   { 4, 4 },
                                   { 5, 5 },
                                   { 6, 6 },
                                   { 7, 7 },
                                   { 8, 8 },
                                   { 9, 9 } };
  
  std::cout << "STD\n";
  for (const auto v : container | std::views::values)
  {
    std::cout << v << "\n";
  }

  etl::map<int, int, 10> container2 = { { 0, 0 },
                                        { 1, 1 },
                                        { 2, 2 },
                                        { 3, 3 },
                                        { 4, 4 },
                                        { 5, 5 },
                                        { 6, 6 },
                                        { 7, 7 },
                                        { 8, 8 },
                                        { 9, 9 } };
  
  std::cout << "\nETL\n";
  for (const auto v : container2 | std::views::values)
  {
    std::cout << v << "\n";
  }
}
```

shreyasbharath, 2021-05-17:
@jwellbelove Which compiler and what version did you use to compile the above example? I tried it out using ARM GCC and GCC 10.1.

jwellbelove, 2021-05-17:
I've successfully compiled on MSVC 2019 and clang v11.0.0 on Windows.

I don't have a version of GCC that seems to support ranges, whether it be on Windows, WSL or a Ubuntu VM. 
Either it says that <ranges> is not found (GCC 9), or says that `std::ranges `or `std::views` 'have not been declared' (GCC 10)

jwellbelove, 2021-05-17:
OK, I managed to install GCC 10.3 to my WSL and compiled my example with `g++ -std=c++20 Ranges.cpp`.
It compiled without errors.

jwellbelove, 2021-05-17:
When run it gives the correct output...
```
STD
0
1
2
3
4
5
6
7
8
9

ETL
0
1
2
3
4
5
6
7
8
9
```

closed 2021-09-10 by jwellbelove