ETLCPP/etl #377: GCC 11 compiler error on circular_buffer

shreyasbharath, 2021-05-17:
Running into the below error when #including `etl/circular_buffer.h`. Any ideas @jwellbelove ?

```
../../middleware/etl/include/etl/circular_buffer.h:987:32: error: expected ')' before 'pbuffer_'
  987 |     icircular_buffer<T>(pointer pbuffer_, size_type max_length)
```

jwellbelove, 2021-05-17:
What version are you using?
In the latest code, line 987 is a comment.

The line in question appears to have a redundant template parameter 
`icircular_buffer<T>(pointer pbuffer_, size_type max_length)`

Does changing the line to `icircular_buffer(pointer pbuffer_, size_type max_length)` make a difference?

matvq9, 2021-05-18:
I have the version `gcc (GCC) 11.1.0` and I have the same issue.

Removing the `<T>` does fix the issue.

jwellbelove, 2021-05-18:
That's great, thanks.

jwellbelove, 2021-05-18:
Fixed v20.10.4

shreyasbharath, 2021-05-19:
Thanks @jwellbelove 😍