ETLCPP/etl #382: Using ETL on Win32 / VS2017

andrey-gvrd, 2021-05-25:
Can this library be used on Win32 / VS2017?

Including ETL in an empty VS2017 project (with a single ELT include) gives lots build errors:
```
Description	File	Line	Suppression State
'add_const': is not a member of 'std'	etl_home\etl\type_traits.h	869	
'add_cv': is not a member of 'std'	etl_home\etl\type_traits.h	919	
'add_lvalue_reference': is not a member of 'std'	etl_home\etl\type_traits.h	1295	
'add_pointer': is not a member of 'std'	etl_home\etl\type_traits.h	839	
'add_volatile': is not a member of 'std'	etl_home\etl\type_traits.h	899	
'constexpr' is not valid here	etl_home\etl\binary.h	104	
'decay': is not a member of 'std'	etl_home\etl\type_traits.h	1267	
'etl::add_const': use of class template requires template argument list	etl_home\etl\type_traits.h	869	
'etl::add_cv': use of class template requires template argument list	etl_home\etl\type_traits.h	919	
'etl::add_lvalue_reference': use of class template requires template argument list	etl_home\etl\type_traits.h	1295	
'etl::add_pointer': use of class template requires template argument list	etl_home\etl\type_traits.h	839	
'etl::add_volatile': use of class template requires template argument list	etl_home\etl\type_traits.h	899	
'etl::decay': use of class template requires template argument list	etl_home\etl\type_traits.h	1267	
'etl::extent': use of class template requires template argument list	etl_home\etl\type_traits.h	1227	
'etl::is_array': use of class template requires template argument list	etl_home\etl\type_traits.h	1019	
'etl::is_base_of': use of class template requires template argument list	etl_home\etl\type_traits.h	1277	
'etl::is_class': use of class template requires template argument list	etl_home\etl\type_traits.h	1286	
'etl::is_compound': use of class template requires template argument list	etl_home\etl\type_traits.h	1009	
'etl::is_const': use of class template requires template argument list	etl_home\etl\type_traits.h	849	
'etl::is_fundamental': use of class template requires template argument list	etl_home\etl\type_traits.h	999	
'etl::is_lvalue_reference': use of class template requires template argument list	etl_home\etl\type_traits.h	1049	
'etl::is_pod': use of class template requires template argument list	etl_home\etl\type_traits.h	1139	
'etl::is_pod': use of class template requires template argument list	etl_home\etl\type_traits.h	1149	
'etl::is_pod': use of class template requires template argument list	etl_home\etl\type_traits.h	1159	
'etl::is_pod': use of class template requires template argument list	etl_home\etl\type_traits.h	1169	
'etl::is_pod': use of class template requires template argument list	etl_home\etl\type_traits.h	1179	
'etl::is_pointer': use of class template requires template argument list	etl_home\etl\type_traits.h	1029	
'etl::is_reference': use of class template requires template argument list	etl_home\etl\type_traits.h	1039	
'etl::is_signed': use of class template requires template argument list	etl_home\etl\type_traits.h	939	
'etl::is_unsigned': use of class template requires template argument list	etl_home\etl\type_traits.h	949	
'etl::is_void': use of class template requires template argument list	etl_home\etl\type_traits.h	979	
'etl::is_volatile': use of class template requires template argument list	etl_home\etl\type_traits.h	879	
'etl::make_signed': use of class template requires template argument list	etl_home\etl\type_traits.h	1196	
'etl::make_unsigned': use of class template requires template argument list	etl_home\etl\type_traits.h	1206	
'etl::negation': use of class template requires template argument list	etl_home\etl\type_traits.h	808	
'etl::rank': use of class template requires template argument list	etl_home\etl\type_traits.h	1257	
'etl::remove_all_extents': use of class template requires template argument list	etl_home\etl\type_traits.h	1247	
'etl::remove_extent': use of class template requires template argument list	etl_home\etl\type_traits.h	1237	
'etl::remove_pointer': use of class template requires template argument list	etl_home\etl\type_traits.h	829	
'explicit' is not allowed	msvc_home\MSVC\14.16.27023\include\yvals.h	361	
'extent': is not a member of 'std'	etl_home\etl\type_traits.h	1227	
'is_array': is not a member of 'std'	etl_home\etl\type_traits.h	1019	
'is_base_of': is not a member of 'std'	etl_home\etl\type_traits.h	1277	
'is_class': is not a member of 'std'	etl_home\etl\type_traits.h	1286	
'is_compound': is not a member of 'std'	etl_home\etl\type_traits.h	1009	
'is_const': is not a member of 'std'	etl_home\etl\type_traits.h	849	
'is_fundamental': is not a member of 'std'	etl_home\etl\type_traits.h	999	
'is_lvalue_reference': is not a member of 'std'	etl_home\etl\type_traits.h	1049	
'is_pod': is not a member of 'std'	etl_home\etl\type_traits.h	1139	
'is_pod': is not a member of 'std'	etl_home\etl\type_traits.h	1149	
'is_pod': is not a member of 'std'	etl_home\etl\type_traits.h	1159	
'is_pod': is not a member of 'std'	etl_home\etl\type_traits.h	1169	
'is_pod': is not a member of 'std'	etl_home\etl\type_traits.h	1179	
'is_pointer': is not a member of 'std'	etl_home\etl\type_traits.h	1029	
'is_reference': is not a member of 'std'	etl_home\etl\type_traits.h	1039	
'is_signed': is not a member of 'std'	etl_home\etl\type_traits.h	939	
'is_standard_layout': is not a member of 'std'	etl_home\etl\type_traits.h	1072	
'is_standard_layout': undeclared identifier	etl_home\etl\type_traits.h	1072	
'is_trivial': is not a member of 'std'	etl_home\etl\type_traits.h	1072	
'is_unsigned': is not a member of 'std'	etl_home\etl\type_traits.h	949	
'is_void': is not a member of 'std'	etl_home\etl\type_traits.h	979	
'is_volatile': is not a member of 'std'	etl_home\etl\type_traits.h	879	
'make_signed': is not a member of 'std'	etl_home\etl\type_traits.h	1196	
'make_unsigned': is not a member of 'std'	etl_home\etl\type_traits.h	1206	
'negation': is not a member of 'std'	etl_home\etl\type_traits.h	808	
[... 359 lines cut ...]
too few parameters for this operator function	etl_home\etl\basic_string.h	2498	
too few parameters for this operator function	etl_home\etl\basic_string.h	2525	
too few parameters for this operator function	etl_home\etl\basic_string.h	2538	
too few parameters for this operator function	etl_home\etl\basic_string.h	2565	
too few parameters for this operator function	etl_home\etl\basic_string.h	2578	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	176	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	178	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	183	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	183	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	190	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	190	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	234	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	236	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	241	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	241	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	248	
variable "uint32_t" is not a type name	etl_home\etl\fnv_1.h	248	
__thiscall can only appear on nonstatic member function declarations	msvc_home\MSVC\14.16.27023\include\yvals.h	361	
```

I'm using `/std:c++17` and `ETL_CPP11_SUPPORTED`, `ETL_CPP14_SUPPORTED`, `ETL_CPP17_SUPPORTED` all evaluate to 1.

Configuring with and without `NO_STL` doesn't change the errors much.

`etl_home` and `msvc_home` are shorthands for actual locations.

jwellbelove, 2021-05-25:
How have you configured the 'Additional includes' in the project settings?
Normally it would be set to point to the ETL's "include" subdirectory, so that ETL headers are included in the form #include "etl/vector.h"

andrey-gvrd, 2021-05-26:
I did.

jwellbelove, 2021-05-26:
Can you post the code you're using or attach your zipped project?

I done all development of the ETL with Visual Studio from VS2013 through to VS2019.

andrey-gvrd, 2021-05-26:
Sure, here it is.
[ETLtest2.zip](https://github.com/ETLCPP/etl/files/6547975/ETLtest2.zip)

jwellbelove, 2021-05-26:
It looks like you are compiling for an SDK that doesn't exist.
Check the  "General Properties" dialog.
[image]

jwellbelove, 2021-05-26:
BTW, I wouldn't normally recommend putting the ETL in Program Files.
I would usually expect to see it in a user folder.

Also it is recommended to set the path to the ETL without the last `/etl`. i.e. The last folder in the path should be `/include`
This allows you to include an ETL header as `#include "etl/array.h"`. This allows the compiler to differentiate between ETL and system headers with the same name. 

```
#include "etl/array.h"

int main() {
	return 0;
}
```

andrey-gvrd, 2021-05-26:
You're right, it was the include path. I had `C:\Program Files (x86)\etl\include\etl` instead of `C:\Program Files (x86)\etl\include`.

Thank you, John.

closed 2021-05-26 by andrey-gvrd