ETLCPP/etl #384: Feature Request etl::message virtual inheritance

fermentedfly, 2021-05-26:
Hi,

I'm currently using some message objects which inherit from etl::message and my own interface.
I'd like my interface to inherit from etl::imessage so i do not need to store references to etl::imessage and my own interface separately.
```c++

class IMyInterface : public etl::imessage
{
virtual void foo() = 0;
};

class MyMessage : public etl::message<...>, public IMyInterface
{
};
``` 
Currently, this does not work since it will result in a diamond like inheritance.
Afaik, virtual inheritance would solve the issue, i.e.

```c++
  template <etl::message_id_t ID_>
  class message : public virtual imessage
```
Would it be possible to change etl::message to use virtual inheritance?

jwellbelove, 2021-06-30:
Would `IMyInterface` be the same for all messages?
(Sorry for the delay in replying)

fermentedfly, 2021-06-30:
No worries, Yes `IMyInterface` is the same for all messages (for a given event bus, there are other buses with with messages not inheriting from `IMyInterface`)

jwellbelove, 2021-06-30:
One option is to add a macro to control this so that it is optional, something like...
`ETL_MESSAGE_INHERITANCE_IS_VIRTUAL`.

Another would be to allow `etl::message` to take another template parameter, which defines what `etl::message` derives from.
This would turn it from multiple inheritance to single inheritance.

jwellbelove, 2021-06-30:
```

       etl::imessage
             |
      ________________
      |              |
etl::message<>  IMyInterface
      |              |
      ________________
             |
          MyMessage

       etl::imessage
             |
        IMyInterface
             |
       etl::message<>
             |
          MyMessage
```

jwellbelove, 2021-06-30:
```
class IMyInterface : public etl::imessage
{
    virtual void foo() = 0;
};

class MyMessage : public etl::message<ID, IMyInterface>
{
    void foo() override;
};
```

fermentedfly, 2021-06-30:
Thanks for the input, if you want me to i will look into it and send a PR if it works out. I guess a second template parameter which defaults to etl::imessage is neater than the using a macro

fermentedfly, 2021-06-30:
something like
```
  template <etl::message_id_t ID_, class PARENT_ = etl::imessage>
  class message : public PARENT_
```

jwellbelove, 2021-06-30:
Pretty much what I was looking at, except I would prefer

```
  template <etl::message_id_t ID_, typename TParent = etl::imessage>
  class message : public TParent
```
An `ETL_STATIC_ASSERT` to check that `etl::imessage` is a base of `TParent` would stop mistakes when using an incompatible type.

jwellbelove, 2021-06-30:
`ETL_STATIC_ASSERT(etl::is_base_of<etl::imessage, TParent>::value, "TParent is not derived from etl::imessage");`

fermentedfly, 2021-06-30:
Ok, will do.

closed 2021-06-30 by fermentedfly