ETLCPP/etl #385: etl::fsm documentation example inconsistencies

gabbla, 2021-06-07:
Hello John,

in the [documentation](https://www.etlcpp.com/fsm.html) of `etl::fsm` (version 20.0.0 and above) there are a couple of inconsistencies:
- `on_event` and `on_event_unknown` APIs take a single parameter, while in the example code them are implemented with 2;
- `on_exit_state` returns `void` (as defined [here](https://github.com/ETLCPP/etl/blob/master/include/etl/fsm.h#L261)) while in the example code implementation it returns an `etl::fsm_state_id_t`.

I am currently on `20.10.6` but the master has the same signatures.

Have a nice day,
Gab

jwellbelove, 2021-06-07:
Thanks for the info. I'll update the docs as soon as I can.

gabbla, 2021-06-07:
You are welcome, thank you.

Let me add that in the first case, the API documentation presents the signatures with a single parameter, but still document the missing one (`sender`) below.

jwellbelove, 2021-06-10:
Done

closed 2021-06-10 by jwellbelove