ETLCPP/etl #391: etl::vector with etl::string not working

fabssor, 2021-06-21:
Hello, 

I have the following problem.

This code sample works:

```
etl::vector<int, 2> test;

test.repair();
```
If I use an `etl::string` as datatype in a vector things break. For example this:

```
etl::vector<etl::string<50>, 2> test;

test.repair();
```
Fails with: terminate called after throwing an instance of 'etl::vector_incompatible_type'

Is it not possible to use etl::string as datatype of etl::vector?

Thanks!

jwellbelove, 2021-06-21:
repair() is a low level function that is only valid for trivially constructable types, which etl::string is not, which is why you are getting the error.

fabssor, 2021-06-22:
So is it illegal to memcpy then a etl::vector with a etl::string?

jwellbelove, 2021-06-22:
A non-trivially constructible object cannot be copied using `memcpy`, as it may result in _undefined behaviour_, due to the object's copy constructor not being invoked.

Using functions such as memcpy, memmove, memclr etc on non-trivially constructible objects can cause all sorts of weird bugs to occur. 
I had to track one down once, when a colleague had decided that the best way to clear an object was `memclr`. It worked fine until a `std::string` was added as a member...

fabssor, 2021-06-24:
Alright Thank you! I will close this then.

closed 2021-06-24 by fabssor