ETLCPP/etl #393: arduino template library: name of the .h file at the root of the src

jerabaul29, 2021-06-26:
I installed the latest ETL with the library manager of the arduino 1.8 IDE. It looks like:

```
jj@MBX20 ~/Arduino/libraries/Embedded_Template_Library_ETL/src $ ls
Embedded_Template_Library.h  etl
```

I wonder if the ```Embedded_Template_Library.h``` is a bit long name. In addition, to import ETL sutff such as ```vector.h```, this means my sketches have to look like:

```
#include "Embedded_Template_Library.h"
#include "etl/vector.h"
```

Would it maybe be more "logical" to call this header for ```etl.h``` so that things look like:

```
#include "etl.h"
#include "etl/vector.h"
```

?

jerabaul29, 2021-06-26:
(note: for one or another reason, I do need to import the first line if any import from the ```etl``` folder is to work out).

jwellbelove, 2021-06-26:
I chose the long header name so that there was no chance that it would clash with any other Arduino library that defined a header called `etl.h`.

jerabaul29, 2021-06-26:
Ok! Sounds reasonable I guess, but maybe a bit confusing (at least it was for me the first time :) ). I guess if #394 gets into something that may help on this point though :) .

closed 2021-06-29 by jwellbelove