ETLCPP/etl #394: arduino embedded template library: add some examples?

jerabaul29, 2021-06-26:
When installing ETL in the arduino IDE 1.8, there are no "examples" available in "File > Examples > Examples from other libraries". I wonder if it may be helpful for "newbies" to have a few basic examples there. Do you think this could be added? I would be willing to contribute a few simple examples if there is a specific "arduino IDE examples" folder in this repo that could be pushed into Arduino IDE examples, if you think this would be a good idea :) .

jwellbelove, 2021-06-26:
That sounds like a good idea. If you want to do a contribute any then that would be great.

jerabaul29, 2021-06-26:
Sounds good!

How do you want that I proceed on this? :)

## setting up examples folder on the arduino repo

Based from how other arduino library repos are set up (taking the example of https://github.com/sparkfun/SparkFun_IridiumSBD_I2C_Arduino_Library ), all what is needed should be to add an ```examples``` folder to the repo https://github.com/ETLCPP/etl-arduino , right? :)

## script for pushing the examples

Based from looking at your repo(s), it looks like all content should live on this repo, while the "arduino" repo is automatically pushed to, right?

I guess this file:

https://github.com/ETLCPP/etl/blob/master/arduino/create_arduino_library.py

would need an additional "section" like "copy arduino examples", or something in this kind? Right?

## location on this repo

Where would you want to have the examples on this repo? Should I add an ```examples-arduino``` folder on this repo for example?

# plan for adding examples

If you agree with these points (or if you have comments, I can change how I plan on doing things), I could just push a simple setup with  a simple arduino-example before starting to fill things up maybe? :) .

jwellbelove, 2021-06-26:
That sounds right.

jerabaul29, 2021-06-27:
Thanks for merging :) If you can update the Arduino etl repo that would be amazing - I would then pull the latest generated arduino library and check the examples one more time / check that they are well available in the Arduino IDE etc :) .

closed 2021-06-29 by jwellbelove