ETLCPP/etl #396: functions to print containers

jerabaul29, 2021-06-26:
If I am right, there are no dedicated functions for printing containers. In arduino, I end up coding again and again some simple functions to print the content of for example my vectors. Do you think there could be a way to integrate such functions on the repo somehow? I understand that this may be a bit "touchy", as for example on arduino the printing function resorts on some form of Serial.print, which may not be standard / available on other SDKs. But still, wondering if there may be a reasonable way to provide the user with some printing functions.

jwellbelove, 2021-06-27:
It may be difficult to create a truly universal print function.
You could create your own generic print function using templates.

e.g.
```
template <typename TIterator>
void Print(const char* prefix, const char* suffix, TIterator first, TIterator last)
{
    while (first != last)
    {
        Serial.print(prefix);
        Serial.print(*first++);
        Serial.print(suffix);
    }
}
```
```
etl::vector<int, 10> v;

Print("Value = ", "\n", v.begin(), v.end());
```

jerabaul29, 2021-06-27:
I agree that it is a bit tough, but still, I think that many users write and rewrite these kind of functions all the time, so may be nice to provide them to the user - and then if something more specific is needed, this is the users responsibility. What do you think?

jwellbelove, 2021-06-27:
I'll give it some thought.
To handle printing structures and classes, as well as fundamental types, it would have to require the definition of an overloaded object-to-string function.

```
struct Point
{
    int x;
    int y;
};

// User defined
void arduino_print(const Point p)
{
    Serial.print(p.x);
    Serial.print(",");
    Serial.print(p.y);
}
```

```
// ETL pre-defined
void arduino_print(const char* text)
{
    Serial.print(text);
}

template <typename TPrefix, typename TSuffix, typename TIterator>
void Print(const TPrefix& prefix, const TSuffix& suffix, TIterator first, TIterator last)
{
    while (first != last)
    {
        arduino_print(prefix);
        arduino_print(*first++);
        arduino_print(suffix);
    }
}
```

```
etl::vector<Point, 10> v;

Print("Value = ", "\n", v.begin(), v.end());
```

jerabaul29, 2021-06-27:
Just brainstorming (not tested in implementation yet, would like to hear what you think before I may start really working on it), and this may be naive thoughts of course, but what would you think about using something like:

- a series macro-defined around the "print functions" to apply for printing individual elements of a container; for example, by default, in arduino ecosystem:

```
#define SERIAL_FOR_PRINT Serial
#define SERIAL_PRINT(x) SERIAL_FOR_PRINT.print(x);
#define SERIAL_PRINTLN(x) SERIAL_FOR_PRINT.println(x);
````

The idea having a macro (with a default value in it) would be that, if the user wants to use something non typical for whatever reason (for example, the serial to use is ```Serial1``` instead of ```Serial```), the user can over-write the corresponding macro(s).

- assuming that, for any type ```T``` for which a container on type ```T``` is defined, there exists a ```Serial.print(T const & element)``` function. That would be automatically satisfied for integral types. For non integral types, this would be the responsibility of the user to define it, by inherinting from the ```Printable``` class (see https://arduino.stackexchange.com/questions/53732/is-it-possible-to-print-a-custom-object-by-passing-it-to-serial-print ; I have not tested it, but according to the discussion there should be straightforward).

- then it would be enough to have a function in the kind you mention. Wondering if it may be useful to have container specific printers too, to show a bit more information. For example, for a vector, something like:

```
template<typename T>
void Print(T const & vec_in){
  SERIAL_PRINT(F("vector has | size "));
  SERIAL_PRINT(vec_in.size());
  SERIAL_PRINT(F(" | capacity "));
  SERIAL_PRINT(vec_in.capacity());
  SERIAL_PRINT(F(" | content "));
  for (T const & elem : vec_in){
    SERIAL_PRINT(elem):
    SERIAL_PRINT(F(" | "));
  }
  SERIAL_PRINTLN();
}
```

Any thoughts on this? :)