ETLCPP/etl #399: type_traits import error

jerabaul29, 2021-06-30:
On the latest Arduino IDE, I am getting the following error:

```
In file included from /home/jr/Arduino/libraries/Embedded_Template_Library_ETL/src/etl/algorithm.h:43:0,
                 from /home/jr/Arduino/libraries/Embedded_Template_Library_ETL/src/etl/vector.h:41,
                 from /home/jr/Arduino/libraries/Embedded_Template_Library_ETL/examples/Vector_Examples/Example_Vector_1_simple_use/Example_Vector_1_simple_use.ino:2:
/home/jr/Arduino/libraries/Embedded_Template_Library_ETL/src/etl/type_traits.h:69:12: fatal error: type_traits: No such file or directory
   #include <type_traits>
            ^~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Uno.
```

Seems to come from:

https://github.com/ETLCPP/etl/blob/7933e146fcd06ace3a4d21d39bb09c4ce4d34700/include/etl/type_traits.h#L68-L70

Any idea how to fix? It is true that there is no ```type_traits``` file.

jwellbelove, 2021-06-30:
Have you tried defining ETL_NO_STL

jerabaul29, 2021-06-30:
You are right, this fixes things. I will add it to the default arduino example.

jerabaul29, 2021-06-30:
Closing this, we can discuss on the pull request if needed.

I think this is a nice illustration of the value of having Arduino IDE examples :) . I was using a much older version of the ETL or something like this so had no problem when I tested a few days ago the Arduino example, but tried again today and this popped up. I think this will help people starting with the library quite a lot in the long run and if enough stuff can get documented :) .

closed 2021-06-30 by jerabaul29