ETLCPP/etl #403: can't create delegate from function pointer

diggit, 2021-07-08:
Hi,
is it possible to create `etl::delegate` instance from function pointer?

```cpp
#include "etl/delegate.h"
using fptr = etl::delegate<void(int, int)>;

volatile int dump;
void foo(int a, int b) {
	dump = a;
	dump = b;
}

int main() {

	auto f = fptr::create<foo>();
	f(1,2);

	auto pf = foo;
	f = fptr::create(pf); // fails to create
	//f = fptr::create<pf>();

	return 0;
}
```

here is live code https://godbolt.org/z/Kcz4Ycn51

jwellbelove, 2021-07-09:
There are two issues in trying to implement what you are asking for.

1. The `etl::delegate` stores an internal pointer to an object, in a `void*`. Storing a pointer to a function in a `void*` is undefined behaviour. To get around this would mean that the delegate would have to store _two_ pointers in its internal invocation structure, thereby increasing delegate storage by the size of a pointer.
2. Defining a uint big enough to hold both an object and function pointer would work, but it would mean that none of the `etl::delegate` functions could be `constexpr`.

The only acceptable solution would be 1.

A workaround would be to wrap the function pointer in a class that is passed to the delegate.

```
void foo(int a, int b) 
{
  dump = a;
  dump = b;
}

struct function_wrapper
 {
   void operator()(int a, int b)
   {
     function_ptr(a, b);
   }

   void (*function_ptr)(int, int);
};

auto pf = foo;

// Make the wrapper
function_wrapper wrapper = { pf };

// Create the delegate.
fptr f(wrapper);

// Call the delegate
fptr(1, 2);
```

In fact, I think I shall add a generic templated wrapper to `etl::delegate`.

jwellbelove, 2021-07-09:
Note: The workaround does require that the function wrapper lifetime is at least as long as the delegate.

jwellbelove, 2021-07-09:
It could also work if you found a way to make your function pointer a constexpr.

jwellbelove, 2021-09-07:
I think I shall stick with the current code and require the wrapper solution..

closed 2021-09-07 by jwellbelove