ETLCPP/etl #405: ETL sets gcc  _LITTLE_ENDIAN and _BIG_ENDIAN flags at the same time

Revoltowiec, 2021-07-26:
Hi, 

During development I came into the problem, that ETL library is setting global GCC _LITTLE_ENDIAN and _BIG_ENDIAN flags at the same time, which breaks rest of my code.

I am using arm-none-abi-gcc / g++ version 9.3.1, for arm cortex mcu in big endian mode.

I found that if there is any include from ETL library, there both of this flags is set.
However when I define ETL_NO_STL flag in configuration file, problems disappears.

Please let me know is there any option to force endianess used by ETL library?

Below steps to reproduce with Compiler Explorer (https://godbolt.org/)

set language to: C++ 
set compiler to: ARM gcc 9.2.1 (none) or 10.2.1 (none), on (linux) versions this problem doesn't appear.
set compiler settings: -std=gnu++2a or -std=gnu++17 (the same behavior)

```
//ETL CONFIGURATION
#define ETL_TARGET_DEVICE_ARM
#define ETL_TARGET_OS_NONE
#define ETL_CPP11_SUPPORTED 1
#define ETL_CPP14_SUPPORTED 1
#define ETL_CPP17_SUPPORTED 1
#define ETL_NO_NULLPTR_SUPPORT 0
#define ETL_NO_LARGE_CHAR_SUPPORT 0
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED 1
#define ETL_ARRAY_VIEW_IS_MUTABLE
//#define ETL_NO_STL //IF UNCOMMENTED PROBLEM DISAPPEARS

//IF ANY UNCOMMENTED PROBLEM APPEARS
// #include "etl/map.h"
// #include "etl/array.h"
#include "etl/vector.h"

#ifdef _LITTLE_ENDIAN
    #ifdef _BIG_ENDIAN
        #error "BOTH ENDIANESS SELECTED"
    #endif
#endif

int main()
{
    return 0;
}

```

jwellbelove, 2021-07-26:
According to what I can find on GCC's endian macros, you should be using these.
```
__BYTE_ORDER__
__ORDER_LITTLE_ENDIAN__
__ORDER_BIG_ENDIAN__
```

If little endian, then `__BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__`
If big endian, then `__BYTE_ORDER__ == __ORDER_BIG_ENDIAN__`

Revoltowiec, 2021-07-27:
You're right. I missed mention it in description, but I have file with:

```
#ifndef CGT_GCC_H_
#define CGT_GCC_H_

#if (__BYTE_ORDER__ == __ORDER_BIG_ENDIAN__)     /* is big endian compile */
  #if !defined(_BIG_ENDIAN)
      #define _BIG_ENDIAN                          /* FAPI generic define for big endian */
  #endif
#endif
#if  (__BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__) /* is little endian compile */
  #if !defined(_LITTLE_ENDIAN)
      #define _LITTLE_ENDIAN                       /* FAPI generic define for little endian */
  #endif
#endif

/* Defines the method to indicate packed enums */
#define ATTRIBUTE_PACKED    __attribute__((packed))

#endif /* CGT_GCC_H_ */
``` 

Which is included to second file, which checks those flags and then define some data structures, based on that.

```
typedef union 
{		
   #ifdef _BIG_ENDIAN
	 struct {} struct_type; 
   #endif
   #ifdef _LITTLE_ENDIAN
	 struct{}struct_type;
   #endif
}union_type;
```

It's a part of mcu vendor C library.

After your suggestion, I checked one more idea -> created wrappers for this library and forced C linkage for all components that use these flags internally and it seems to work. 

Thanks for help and this issue could be closed.

closed 2021-07-27 by jwellbelove