ETLCPP/etl #406: etl::is_base_of type trait fails after enabling ETL_NO_STL flag

Revoltowiec, 2021-07-26:
Hi, 

After enabling ETL_NO_STL flag, I get a problem with __etl::is_base_of__ type trait.

It uses __check__ function under the hood, which seems to be a problem.

Compiler used arm-none-eabi-gcc / g++ version 9.3.1

```
FAILED: ...
In file included from ../utils/src\assertion.hpp:14:
In file included from ../etl/include\etl/algorithm.h:43:
../etl/include\etl/type_traits.h:642:22: error: base specifier must name a class
    struct internal: TDerived, dummy<int>{};
                     ^~~~~~~~
../etl/include\etl/type_traits.h:649:39: note: in instantiation of member class 'etl::is_base_of<unsigned char [10], unsigned char [10], false>::internal' requested here
    static const bool value = (sizeof(check((internal*)0)) == sizeof(TBase*));
                                      ^
../etl/include\etl/pool.h:159:60: note: in instantiation of template class 'etl::is_base_of<unsigned char [10], unsigned char [10], false>' requested here
      ETL_STATIC_ASSERT((etl::is_same<U, T>::value || etl::is_base_of<U, T>::value), "Pool does not contain this type");
                                                           ^
../io_framework/test/unit_tests/../../../io_framework\src/memory_pool.hpp:99:15: note: in instantiation of function template specialization 'etl::pool<unsigned char [10], 3>::release<unsigned char [10]>' 
requested here
      memPool.release(elementPtr);
              ^

... some lines cut for readability  ...

In file included from ../etl/include\etl/algorithm.h:43:
../etl/include\etl/type_traits.h:649:39: warning: use of member 'check' before its declaration is a Microsoft extension [-Wmicrosoft-template]
    static const bool value = (sizeof(check((internal*)0)) == sizeof(TBase*));
                                      ^
../etl/include\etl/pool.h:159:60: note: in instantiation of template class 'etl::is_base_of<unsigned char [10], unsigned char [10], false>' requested here
      ETL_STATIC_ASSERT((etl::is_same<U, T>::value || etl::is_base_of<U, T>::value), "Pool does not contain this type");
                                                           ^
../io_framework/test/unit_tests/../../../io_framework\src/memory_pool.hpp:99:15: note: in instantiation of function template specialization 'etl::pool<unsigned char [10], 3>::release<unsigned char [10]>' 
requested here
      memPool.release(elementPtr);
              ^

... some lines cut for readability  ...

../etl/include\etl/type_traits.h:644:19: note: member is declared here:
    static TBase* check(TBase*);
                  ^
../etl/include\etl/type_traits.h:645:38: note: member is declared here
    template<typename T> static char check(dummy<T>*);
                                     ^
../etl/include\etl/type_traits.h:649:39: error: no matching function for call to 'check'
    static const bool value = (sizeof(check((internal*)0)) == sizeof(TBase*));
                                      ^~~~~
../etl/include\etl/type_traits.h:644:19: note: candidate function not viable: no known conversion from 'etl::is_base_of<unsigned char [10], unsigned char [10], false>::internal *' to 'unsigned char *[10]' for 1st argument
    static TBase* check(TBase*);
                  ^
../etl/include\etl/type_traits.h:645:38: note: candidate template ignored: could not match 'dummy<type-parameter-0-0>' against 'etl::is_base_of<unsigned char [10], unsigned char [10], false>::internal'   
    template<typename T> static char check(dummy<T>*);
                                     ^
1 warning and 2 errors generated.

```

Steps to reproduce in compiler explorer:
set language to: C++
set compiler to: ARM gcc 9.2.1 (none)
set compiler settings: -std=gnu++2a or -std=gnu++17 (the same behavior)
```
//ETL CONFIGURATION
#define ETL_TARGET_DEVICE_ARM
#define ETL_TARGET_OS_NONE
#define ETL_CPP11_SUPPORTED 1
#define ETL_CPP14_SUPPORTED 1
#define ETL_CPP17_SUPPORTED 1
#define ETL_NO_NULLPTR_SUPPORT 0
#define ETL_NO_LARGE_CHAR_SUPPORT 0
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED 1
#define ETL_ARRAY_VIEW_IS_MUTABLE
#define ETL_NO_STL

#include "etl/pool.h"

using elem_type = uint8_t[10];
int main()
{
    etl::pool<elem_type, 3> memPool {};

    elem_type* memory = memPool.allocate();
    memPool.release(memory);
    return 0;
}
```

Compiler explorer returns:
```
In file included from /opt/compiler-explorer/libs/etl/include/etl/utility.h:35,
                 from /opt/compiler-explorer/libs/etl/include/etl/ipool.h:38,
                 from /opt/compiler-explorer/libs/etl/include/etl/pool.h:35,
                 from <source>:14:
/opt/compiler-explorer/libs/etl/include/etl/type_traits.h: In instantiation of 'struct etl::is_base_of<unsigned char [10], unsigned char [10], false>::internal':
/opt/compiler-explorer/libs/etl/include/etl/type_traits.h:649:44:   recursively required by substitution of 'template<class T> static char etl::is_base_of<unsigned char [10], unsigned char [10], false>::check<T>(etl::is_base_of<unsigned char [10], unsigned char [10], false>::dummy<T>*) [with T = <missing>]'
/opt/compiler-explorer/libs/etl/include/etl/type_traits.h:649:44:   required from 'const bool etl::is_base_of<unsigned char [10], unsigned char [10], false>::value'
/opt/compiler-explorer/libs/etl/include/etl/pool.h:159:7:   required from 'void etl::pool<T, VSize>::release(const U*) [with U = unsigned char [10]; T = unsigned char [10]; unsigned int VSize = 3]'
<source>:22:27:   required from here
/opt/compiler-explorer/libs/etl/include/etl/type_traits.h:642:12: error: base type 'unsigned char [10]' fails to be a struct or class type
  642 |     struct internal: TDerived, dummy<int>{};
      |            ^~~~~~~~
Compiler returned: 1
```

Potentially regression could be introduced with https://github.com/ETLCPP/etl/issues/371

Expected behavior:
- no new compilation warnings (__warning: use of member 'check' before its declaration is a Microsoft extension [-Wmicrosoft-template]__)
- stable build with ETL_NO_STL flag

jwellbelove, 2021-07-26:
This line in `etl::is_base_of`...
`const bool IsFundamental = (etl::is_fundamental<TBase>::value || etl::is_fundamental<TDerived>::value)>`

should be changed to...
`const bool IsFundamental = (etl::is_fundamental<TBase>::value || etl::is_fundamental<TDerived>::value || etl::is_array<TDerived>::value)>`

Also, try changing the declarations of `check()` to be actual functions to see if it gets rid of the warning.
```
static TBase* check(TBase*) {}
    
template<typename T> 
static char check(dummy<T>*) {}
```

Let me know if that solves your issues.

jwellbelove, 2021-07-26:
Sorry, I left out adding return values for those functions.

Revoltowiec, 2021-07-27:
Replacement of _IsFundamental_ fix my build issue and clear warnings. 

Thanks!

jwellbelove, 2021-07-27:
That's great, I'll add the changes to the next push.

closed 2021-07-27 by jwellbelove