ETLCPP/etl #409: Adding rust-like result conatainer?

skorokhod, 2021-08-08:
Thanks for the Great Library!
What do you think about adding some Rust-like 'result' container (https://doc.rust-lang.org/std/result/)
which can hold one of two value types. It's very convenient to use for functions which should return value or error-code.
Moreover it can be integrated with optional-type to extract the result

jwellbelove, 2021-08-08:
Thanks, I'll take a look.

EvenTang, 2021-08-11:
Good idea.
Actually, [AutoSAR Adaptive](https://www.autosar.org/), which is a standard in automotive industry, adopts the same idea.
_Specification of Core Types for Adaptive Platform_ is a very good reference.
Chapter 7.1 is the error handling philosophy, and 8.3 describes the specification of **Result.**

skorokhod, 2021-08-28:
will it allow to use like Result<void, Error>, Result<int, int>?

jwellbelove, 2021-08-28:
Yes, it would.

jwellbelove, 2021-09-07:
20.17.0
Added `etl::result`

closed 2021-09-07 by jwellbelove