ETLCPP/etl #411: add Reader & Writer conecpts over buffer

skorokhod, 2021-08-11:
It's quite convenient to have some cover over buffer which contains span over buf & write position.
This helps you to read data by chunks in a manner like:
```
uint8_t buf[100];
mem_writer writer(buf);

writer.write<uint8_t>(1); //write 1 byte & move position by 1
writer.write<uint64_t>(0x234234); //write next 8 bytes & move position by 8

//copy data from sapn into buf - write whole buffer into buf and move write position
writer.write(std::span<uint8_t>(/*...*/)); 

// and so on ....

writer.empty(); //check if some data written
writer.size_used(); //get size of written data
writer.data() -> std::span() //written data 
```

And something in opposite side - mem_reader,  wich helps to read data like:
```
//got some paclket from somewhere in span:
std::span<uint8_t> find_packet(std::span<uint8_t>)
{
    mem_reader reader(std::span<uint8_t>());
    if (reader.empty() || reader.size_left() < N) //check,  that buffer contains enought data to decode
        return;

    uint8_t b1 = reader.read<uint8_t>();       //gets 1 byte from reader and moves pointer forward by 1
    uint8_t b2 = reader.read<uint8_t>();       //gets 1 byte from reader and moves pointer forward by 1 more
    uint16_t d3 = reader.read<uint8_t>();
    std::span<uint8_t> data = reader.read_raw_data(16); //get's 16 bytes from reader and moves pointer forward by 16

    //then check the size of data left and make some slice,  for example:
    reader.size_left();
    return reader.data_left();
}
```

An example of some implementation you can get from:
https://github.com/CPB9/bmcl/blob/master/src/bmcl/bits/MemReaderDecl.h
https://github.com/CPB9/bmcl/blob/master/src/bmcl/bits/MemWriterDecl.h

jwellbelove, 2021-08-17:
The concept is very similar to the current `etl::bit_stream`. I think a complementary `etl::byte_stream` would be a good addition that would implement the concepts you are thinking of.

jwellbelove, 2021-08-17:
What about byte order?
The ETL's `bit_stream` assumes network byte order.
Should it mandate network order or be selectable?

skorokhod, 2021-08-17:
> What about byte order?
> The ETL's `bit_stream` assumes network byte order.
> Should it mandate network order or be selectable?

That's funny,  I've thought  about it and opened github to leave a comment now)

So, I used it like a way to serialize data between different platforms and all of them are little endian. Moreover, some data are in raw and I have no way to deserialize it from little endian to put them next in big. I

I think, that etl::byte_stream will be faster and more convenient than bit_stream. It should be little endian first, because of number of little endian platforms.

skorokhod, 2021-08-17:
I took a look at `etl::bit_stream`. I looks like there's no way to know how many free bits left in the buffer. Also its not clear at which order the reading happens. Is it FIFO like bit buffer?

When we choose the way to implement bytes storage,  we decided to strongly  divide reading & writing by class-types to eliminate any ambiguity. Just added methods to get reader from writer and back, like std::span.

And one more question. Why you're not using optional values for:
```
template <typename T>
bool get(T& value, uint_least8_t width = CHAR_BIT * sizeof(T));

```

I think that it will be more clean:
```
template <typename T>
std::optional<T> get(uint_least8_t width = CHAR_BIT * sizeof(T));

```

and than:

```
std::optional<char> c1 = bit_stream.get(c1, 6);
if (c1) ...
//or
if (c1.has_value())...
//or
c1.value_or(0);
```

jwellbelove, 2021-08-17:
`etl::bit_stream` was not meant to be FIFO like. It was designed to encode/decode a buffer of data that had been 'bit compressed' to minimum size. I used to write code to encode/decode serial protocols where the data values would be of arbitrary bit length and could cross byte boundaries. i.e The time in H:M:S would be transmitted as 5:6:6 = 17 bits.

The encoder would use 'put()' to build the bit stream and the decoder, `get()` to extract.

I didn't use `etl::optional` as not having enough bits to extract a value should be an exceptional occurrence; it would mean that either you had not received enough bytes of data (resulting in a bad checksum) or you had tried to extract more values than the protocol spec defined (software bug).

`etl::bit_stream` does have two member functions to examine the number of bytes/bits that have been used so far; `size()` & `bits()`.
It could probably do with specific functions to return the number of available bits/bytes.

jwellbelove, 2021-08-18:
Actually I'm coming over to the idea of using `etl::optional` as it does feel like a more 'modern' way of implementing it.

jwellbelove, 2021-08-23:
I've pushed an implementation of `etl::byte_stream_writer` and `etl::byte_stream_reader` to feature branch `feature/byte_stream`.
Give it a look and let me know you think any changes are required.

jwellbelove, 2021-08-24:
**API**
*******************************************************************************
**Common functions.**

`void restart(size_t n = 0U)`
    Sets the index back to position 'n' in the stream. Default = 0.

`const_iterator begin() const`
    Returns start of the stream.

`const_iterator end() const`
    Returns end of the stream.

`const_iterator cbegin() const`
    Returns start of the stream.

`const_iterator cend() const`
    Returns end of the stream.

*******************************************************************************
**byte_stream_reader**

`byte_stream_writer(char* begin_, char* end_, etl::endian buffer_endianness_ = etl::endian::big)`
    Construct from range.

`byte_stream_writer(char* begin_, size_t length_, etl::endian buffer_endianness_ = etl::endian::big)`
    Construct from begin and length.

`bool write(bool value)`
    Writes a boolean to the stream

`template <typename T>`
`bool write(T value)`
    Write a value to the stream. (integral or floating point)

`template <typename T>`
`bool write(const etl::span<T>& range)`
    Write a range of T to the stream. (integral or floating point)

`bool full() const`
    Returns true if the byte stream index has reached the end.

`bool empty() const`
    Returns true if the byte stream is empty.

`size_t size_bytes() const`
    Returns the number of bytes used in the stream.

`size_t capacity() const`
    Returns the maximum number of bytes in the stream.

`template <typename T>`
`size_t available() const`
    The number of T left in the stream.

*******************************************************************************
**byte_stream_reader**

`byte_stream_reader(char* begin_, char* end_, etl::endian buffer_endianness_ = etl::endian::big)`
    Construct from range.

`byte_stream_reader(char* begin_, size_t length_, etl::endian buffer_endianness_ = etl::endian::big)`
    Construct from begin and length.

`template <typename T>`
`etl::optional<T> read()`
    Read a value from the stream.

`template <typename T>`
`etl::optional<etl::span<T>> read(size_t n)`
    Read a byte range from the stream.

`template <typename T>`
`etl::optional<etl::span<T>> read(etl::span<T> range)`
    Read a range of T from the stream.

`bool empty() const`
    Returns true if the byte stream is empty.

`size_t size_bytes() const`
    Returns the number of bytes used in the stream.

`template <typename T>`
`size_t available() const`
    The number of T left in the stream.

skorokhod, 2021-08-28:
it will be helpful to add for writer:
`fillUp(value, count)`
`span written_data()` - I'm not sure about span, but we need the way to get written part of buffer
`byte_stream_reader written_data()`
`span left_data()`

as for the reader:
`byte_stream_reader data_left()`
`skip(count)`
`copy(count, dest)` - but I'm not sure about the safe way to make it. may be it can be even:
`copy(size_t count, byte_stream_writer)` - but in this case it's not clear what hould we do with endianness

and both of them have to have ctors for c arrays like:
```
template <std::size_t n, typename R>
byte_stream_reader(R (&array)[n])
```

jwellbelove, 2021-08-28:
begin() and end() should span the actual data in the buffer.

etl::span s(stream.begin(), stream.end());

jwellbelove, 2021-09-02:
Does `fillUp(value, count)` update the length of the written buffer, or just pre-initialise the underlying buffer?

jwellbelove, 2021-09-02:
```
copy(count, dest)
copy(size_t count, byte_stream_writer)

```
`read(size_t n)` returns a span from the reader allowing the resulting range to be used in a copy.

jwellbelove, 2021-09-07:
20.17.0
Added `etl::byte_stream`

closed 2021-09-07 by jwellbelove