ETLCPP/etl #420: Version documentation

florianhumblot, 2021-08-30:
Currently looking up documentation for ETL is quite confusing because the website that hosts the docs isn't versioned. This makes  it very hard to look up the docs  for a particular function or class because your project might be a few releases behind. 

Is there any way to access "old" version of the online documentation?
Since it looks like the documentation is a static website it could be as easy as putting the docs in a directory structure like:

```
webroot/
├─ latest/
├─ 20.15.0/
├─ 20.14.1/
├─ 20.14.0/
├─ ...
├─ 19.2.0/
```

Where `latest/` is a symlink to the latest version of the software.
Another option would be a service like readthedocs but I don't know enough about that (or other) platform to endorse it.

A recent confusing example for me:
I am using ETL 18.11.0, and I was looking at the docs for strings and how to use external buffers. 
The current docs speak of functions like `string::initialize_free_space()` and `string::trim_to_terminator()` which weren't introduced until later versions than what I was using. The only way for me to find this out was to dive into the library code itself and see if the functions used exist.

Much easier would be if I could navigate to the docs for my ETL version and read that instead :)

jwellbelove, 2021-08-30:
I'll look into it to see what I can do.

As most of the code does not change from version to version, I'm not keen to replicate the entire site each time, as I may soon run out of web space, so I was thinking that I may version each page; the one you see being the latest and a drop down list for previous versions.

_readthedocs_ looks interesting, but it would require me to spend a lot of time recreating the documentation in the new format. (I don't get paid to do any of this, so I have to do it all in my spare time!)

florianhumblot, 2021-08-30:
versioning the pages sounds interesting, although it may be challenging to keep the links working (i.e. if you're on a non-latest page, say strings:20.0.1, and click on the w_string link, should it take you to the latest version of w_string, or to the version that 20.0.1 was built against or another option).

If you move to a documentation system which has the source of the documentation in a repository you could also enlist the help of contributors to keep it up to date. (I've also just noticed that readthedocs is free for open-source projects, including the hosting of it)  

If there's anything I can contribute to this issue let me know, I'll be glad to contribute to the best of my ability :)

jwellbelove, 2021-08-30:
I quite like the fact that _readthedocs_ based documentation can be downloaded in an offline format such as PDF.

closed 2021-09-04 by jwellbelove